pimoroni/pimoroni-pico #11: MicroPython Firmware none existant

qwerty822, 2021-01-22:
the link to the micropython firmware leads nowhere

veryalien, 2021-01-23:
"setup is a breeze"....
It would be if the uf2 file was anywhere to be found.
The broken link to no file appears on github and your pico product pages, please fix asap.

For the experts this won't be a show-stopper, but for people wanting a nice, simple plug and play experience with a pico explorer or rgb base, it's not good at all.

Gadgetoid, 2021-01-23:
The SDK and MicroPython port were in constant flux prior to release and even now have bugs and issues. Pico is going to have a rocky start I’m afraid. The product pages may be a touch optimistic. We’ll get there.

Yes the alpha release - and future releases - supplied in the release section of this GitHub repository are the firmware you should be using.

lowfatcode, 2021-01-23:
You can find the releases section here which also includes links to download pre-compiled (`.uf2` files) of our MicroPython firmware: https://github.com/pimoroni/pimoroni-pico/releases

closed 2021-01-23 by lowfatcode