pimoroni/pimoroni-pico #9: picodisplay micropython example potential typo

jpwsutton, 2021-01-22:
I think I've found a possible typo in the pico_display micopython example:
https://github.com/pimoroni/pimoroni-pico/blob/f9ad11181d2618e4f45c0b2c84eedef0e07d334b/micropython/examples/pico_display/demo.py#L3

The interpreter complains that `init()` should have an argument:
```
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
TypeError: function takes 1 positional arguments but 0 were given
```

My best guess is that in the module, the init call requires a buffer:https://github.com/pimoroni/pimoroni-pico/blob/f9ad11181d2618e4f45c0b2c84eedef0e07d334b/micropython/modules/pico_display/pico_display.h#L6

This can be seen in the C++ example:

```
uint16_t buffer[PicoDisplay::WIDTH * PicoDisplay::HEIGHT];
PicoDisplay pico_display(buffer);
```

Was found using the 0.0.1 Alpha `firmware.uf2` (https://github.com/pimoroni/pimoroni-pico/releases/tag/v0.0.1)

jpwsutton, 2021-01-22:
Ok!

This is a working example:

```
import time, random
import picodisplay as display

buf = bytearray(240*135)
display.init(buf)

display.set_backlight(1.0)
i = 0
width = display.get_width()
height = display.get_height()

class Ball:
    def __init__(self, x, y, r, dx, dy, pen):
        self.x = x
        self.y = y
        self.r = r
        self.dx = dx
        self.dy = dy
        self.pen = pen

# initialise shapes
balls = []
for i in range(0, 100):
    balls.append(
        Ball(
            random.randint(0, width),
            random.randint(0, height),
            random.randint(0, 10) + 3,
            random.randint(0, 255) / 128,
            random.randint(0, 255) / 128,
            display.create_pen(random.randint(0, 255), random.randint(0, 255), random.randint(0, 255)),            
        )
    )
    
while True:
    display.set_pen(120, 40, 60)    
    display.clear()
    
    
    display.set_pen(display.create_pen(100,100,0))
    display.circle(100,100, 50)
        
    display.update()
    time.sleep(0.01)
```

Note, I had to remove the fancy random ball loop as it couldn't reference any of the attributes of the ball class:
```
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 40, in <module>
AttributeError: 'int' object has no attribute 'x'
```

Gadgetoid, 2021-01-23:
The “bytearray” should be 240*135*2 since it’s a 16-bit display. Will get this tweaked.

We very much hoped to internalise the memory allocation and avoid this little wart but ... MicroPython is *hard* 😬 (or, rather, stopping GC from eating your buffer is)

ZodiusInfuser, 2021-01-23:
As Gadgetoid mentions, we had versions where the bytearray was internal, but firmware got changed and this example slipped through the net. For now I would suggest changing the start of your program to this, to avoid hardcoding the width and height.

```
width = display.get_width()
height = display.get_height()
buf = bytearray(width * height * 2)
display.init(buf)

display.set_backlight(1.0)
i = 0
```

Gadgetoid, 2021-01-23:
Have made these tweaks as of: https://github.com/pimoroni/pimoroni-pico/commit/d66f2203e76bd5acf27321438048d9f6017be822

jpwsutton, 2021-01-23:
Ah! That must explain why I was seeing some very strange behaviour with the while loop!
Thanks @Gadgetoid!

closed 2021-01-23 by lowfatcode