pimoroni/pimoroni-pico #18: Pico Explorer Base - Correct Piezo pin?

Tohaker, 2021-01-24:
I'm trying to get the Piezo to make any sounds, and I started with the example code which specified the following code to set tones:
```cpp
pico_explorer.set_audio_pin(pico_explorer.GP0);
pico_explorer.set_tone(440, 0.5);
```
However on my hardware, nothing is heard when this code runs. I've tried probing the board with my Multimeter, but I can't find where the trace from the piezo leads. Is GP0 the correct pin, or is there another I should use?

Gadgetoid, 2021-01-24:
This confused the heck out of me, too, as I was writing the yet-to-be-merged documentation.

There is no *right* pin, you can output to any of the GPn pins, but then you need to use a jumper wire to bridge that pin to the "AUDIO" pin on the Pico Explorer header.

This is so the peizo can be wired a little more creatively than if it were tied directly to a pin. But it's a heck of a gotcha!

Tohaker, 2021-01-24:
Wow, that is a logical leap! That all works to me now, thank you! I was curious what the `AUDIO` pin was for, but it was omitted from the handy pin list on the back. Perhaps that's something that can be added in a future hardware revision!

closed 2021-01-24 by Tohaker

johnk42, 2021-11-30:
Thank you. This has been irritating me.