pimoroni/pimoroni-pico #20: PicoDisplay

alvinmb, 2021-01-24:
Does any documentation and examples exist for using the Pico display using Micropython?
I have downloaded alpha 0.3 however not sure how much functionality is working. For example have issues with the RGB led and buttons.

Gadgetoid, 2021-01-24:
We're working on it - what was the problem with the RGB led and buttons?

alvinmb, 2021-01-24:
i reviewed the code for the display driver and found some command's that I
tried.

display.set_led(0, 255, 0)
however it does not set the colour of the RGB let. The rgb led remains off.
It does flash red on a power on event

I also could not affect the display backlight if I change the value from 1.0
display.set_backlight(1.0)

The demo.py ( bubbles) only affects the centre part of the display
Again these issues may be due to a misunderstanding of what these commands
should do.

Regards

Alvin Brown

Gadgetoid, 2021-01-24:
Yup missing documentation is entirely at fault here- it's like expecting you to drive a car without turning the engine on!

You must first:

```python
import picodisplay

# Initialise Picodisplay with a bytearray display buffer
buf = bytearray(picodisplay.get_width() * picodisplay.get_height() * 2)
picodisplay.init(buf)
```

`picodisplay.init()` sets up all the PWM, GPIO and whatnot for all the other functions to work.

Though `demo.py` not working seems... suspect!

alvinmb, 2021-01-24:
Hi
just loaded 004 alpha and the display issues appear to be resolved
will check RGB LED next

UnfinishedStuff, 2021-01-24:
I've documented what I can here:

https://github.com/UnfinishedStuff/Pimoroni_Pico_Display_Pack_documentation

@Gadgetoid , if that can be of any use to you feel free to pinch it, or I can try to bash it into shape if you need a particular style.

alvinmb, 2021-01-24:
Much appreciated

Gadgetoid, 2021-01-24:
@UnfinishedStuff ... amazing stuff! I mean I'm genuinely in awe. I've made it about as far as an example program since I've been working on the C++ side of the docs. I think our documentation will be a little lighter- ie: a dry function reference- than your almost comprehensive tutorial but I'd love to draw attention to what you're doing there.

Your use of an image to illustrate gritty things like clip is something I wanted to integrate too.

P.S. if you:

```python
>>> import picodisplay
>>> dir(picodisplay)
['__class__', '__name__', 'clear', 'update', 'BUTTON_A', 'BUTTON_B', 'BUTTON_X', 'BUTTON_Y', 'character', 'circle', 'create_pen', 'get_height', 'get_width', 'init', 'is_pressed', 'pixel', 'pixel_span', 'rectangle', 'remove_clip', 'set_backlight', 'set_clip', 'set_led', 'set_pen', 'text']
>>> 
```

You'll see there are constants defined for `BUTTON_A` etc, that map them to the corresponding numbers.

UnfinishedStuff, 2021-01-24:
@Gadgetoid Thanks, my board arrived yesterday and I was too excited to wait for official docs!

I'll take a look at the constants, thanks for pointing those out.  I skimmed most of that info from `pico_display.cpp` so I'll have missed anything which isn't in there.

callimero, 2021-01-31:
I wonder why there is no line draw command, I think thats maybe more basic than a circle or rectangle?

alvinmb, 2021-01-31:
Good point. I requested the same

andypiper, 2021-02-07:
> display.set_led(0, 255, 0)
> however it does not set the colour of the RGB let. The rgb led remains off.
> It does flash red on a power on event

I found I had to take the spaces out from between the RGB values in order for the `set_led` functions to work.

martintsw, 2021-02-07:
> > display.set_led(0, 255, 0)
> > however it does not set the colour of the RGB let. The rgb led remains off.
> > It does flash red on a power on event
> 
> I found I had to take the spaces out from between the RGB values in order for the `set_led` functions to work.

Very strange, I don't think the spaces _should_ be significant.  The snippet of `set_led` I included above was a copy/paste from source that I tested before posting.  Did you get an error?  It may be interesting to try it again _with_ the spaces to double confirm.

helgibbons, 2021-02-16:
Closing this one as I think all the issues in this thread should be sorted now.

I've linked to @UnfinishedStuff 's guide from the MP Display readme [#61] so hopefully it's easy for folks to find :)

closed 2021-02-16 by helgibbons