pimoroni/pimoroni-pico #23: display size - Alpha 0.5

alvinmb, 2021-01-25:
display.get_height() returns incorrect value 135 and not 240
>> 

>>> import picodisplay as display
>>> width = display.get_width()
>>> width
240
>>> height = display.get_height()
>>> height
135

helgibbons, 2021-01-25:
Assuming you're using a Pico Display Pack, the display is 240x135?

alvinmb, 2021-01-25:
Hi,
Im using the Pico Explorer Base and Micropython

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

helgibbons, 2021-01-25:
Aha! You'll need to use the picoexplorer library to work with the screen on Explorer Base:

```
import picoexplorer as explorer

width = explorer.get_width()
height = explorer.get_height()

display_buffer = bytearray(width * height * 2)  # 2-bytes per pixel (RGB565)
explorer.init(display_buffer)

print(width)
print(height)

explorer.set_pen(255, 0, 0)  # Set a red pen
explorer.clear()             # Clear the display buffer
explorer.update()            # Update the display with our changes
```

alvinmb, 2021-01-25:
Thank you I just arrived at the same conclusion. This has addressed by
display size issue.
Hower the script below still does not exercise the 3 colour LED

import picoexplorer as explorer

width = explorer.get_width()
height = explorer.get_height()

display_buffer = bytearray(width * height * 2)  # 2-bytes per pixel (RGB565)
explorer.init(display_buffer)

print(width)
print(height)

explorer.set_pen(0, 0, 0)  # Set a red pen
explorer.clear()             # Clear the display buffer
explorer.update()

# And now, lets cycle the LED through a few colours
explorer.set_led(255,0,0) # Set the LED to bright red
time.sleep(1) # Wait for 1 second

explorer.set_led(0,255,0) # Set the LED to bright green
time.sleep(1) # Wait for 1 second

explorer.set_led(0,0,255) # Set the LED to bright blue
time.sleep(1) # Wait for 1 second

explorer.set_led(0,0,0) # Turn the LED off# Update the display with our
changes

Regards

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

helgibbons, 2021-01-25:
Sorted :) I've replied to your other issue about the LED.

closed 2021-01-25 by helgibbons