pimoroni/pimoroni-pico #24: rgb-led not working with attached script - Alpha 0.5

alvinmb, 2021-01-25:
running the script below does not change the status of the led. led is always off.

# A complete example for using the LED

# Import a module which will let us wait for a few seconds
import time

#Standard boilerplate code for using the Display Pack
import picodisplay as display
width = display.get_width()
height = display.get_height()
display_buffer = bytearray(width * height * 2)
display.init(display_buffer)
display.clear()
# And now, lets cycle the LED through a few colours
display.set_led(255,0,0) # Set the LED to bright red
time.sleep(1) # Wait for 1 second

display.set_led(0,255,0) # Set the LED to bright green
time.sleep(1) # Wait for 1 second

display.set_led(0,0,255) # Set the LED to bright blue
time.sleep(1) # Wait for 1 second

display.set_led(0,0,0) # Turn the LED off

helgibbons, 2021-01-25:
This code is working fine for me using Thonny and the 0.0.5 firmware. If those titles from the doc are in your code, it might be worth checking you've got a # in front of them to comment them out?

[image]

alvinmb, 2021-01-25:
Thank you for the picture, However I'm not using a pico display but a pico
Explorer Base. The LED will flash red on a power up, so I would say the LED
is most likely OK, but I cant drive it via the software provided, \i have
tried all build from 0.001 alpha to 005 alpha.

Regards

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

helgibbons, 2021-01-25:
I think you're referring to the indicator LED next to MOTORS, is that right? This LED is part of the motor control circuit - it will light up when things connected to the motor terminals are drawing too much current. It's not exposed directly, so you won't (as far I know!) be able to control it with code.

alvinmb, 2021-01-25:
OK thanks.
yes makes sense, I just discovered that the picoexplorer does not have a
set_led command.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2021-01-25 by helgibbons