pimoroni/pimoroni-pico #25: pico_display.text does not render

JackBonnellDevelopment, 2021-01-25:
Looking at the c++ code you have available on this repo I cannot get the display to render text from the pico_display library using:

`pico_display.text("Test",Point(10,5),0);`

There are no errors and it builds but there is no text on the screen.

Pinkyatacme, 2021-02-05:
Are you also including:
`pico_display.update()`
after that - I have been caught out by similar things in the past.

The various commands build the image in memory, but it isn't sent to the display until you call the update function.

martintsw, 2021-02-07:
> Looking at the c++ code you have available on this repo I cannot get the display to render text from the pico_display library using:
> 
> `pico_display.text("Test",Point(10,5),0);`
> 
> There are no errors and it builds but there is no text on the screen.

It is missing a call to set_pen to make it different from the _background_ (_ie_ text_rect).

Try a highly contrasting value initially.

martintsw, 2021-02-07:
> Looking at the c++ code you have available on this repo I cannot get the display to render text from the pico_display library using:
> 
> `pico_display.text("Test",Point(10,5),0);`
> 
> There are no errors and it builds but there is no text on the screen.

The  examples are missing a call to set_pen to make it different from the background (_ie_ text_rect).
(They also have the capitalization of the Rect and Point classes incorrect)

Try a highly contrasting value initially, _eg_

```
pico_display.set_pen( 0, 255, 0 );
pico_display.text("Test",Point(10,5),0);
```

Gadgetoid, 2021-03-31:
Making this as an enhancement because many of our C examples are still a hot mess.

closed 2022-06-27 by Gadgetoid