pimoroni/pimoroni-pico #30: Getting Started MicroPython & pico_rgb_keypad

soneups, 2021-01-27:
I have followed these [instructions](https://github.com/pimoroni/pimoroni-pico/blob/main/setting-up-micropython.md) to prepare/flash the drive for Micropython

I then downloaded the [example](https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/pico_rgb_keypad)  renamed to main.py and copied over to the pico - but the keypad did not illuminate - if i connect direct to the Pico REPL i can put in enough of the code to illuminate the keypads so ruling out my soldering skills! - but was using a chromebook and the beagle terminal app - Is my approach in renaming the example to main.py and copying to the pico a sound approach?

Gary.

Pys3nberg, 2021-01-27:
Hi,

How are you copying/saving main.py to the pico? I have managed to get it working by saving main.py to the device through Thonny - https://thonny.org/ . I am not sure you can just drag and drop python files over to it, i might be wrong.

soneups, 2021-01-27:
Hi @Pys3nberg i only had my chromebook to hand so i dragged and dropped - but now you have pointed this out i wonder why i even thought it would work this way and can only blame lockdown! - will look to try the 'normal' way.

closed 2021-01-27 by soneups