pimoroni/pimoroni-pico #31: USB_HID using pico_rgb_keypad and micropython

Pys3nberg, 2021-01-27:
Hi,

Is there a way to emulate key presses to an OS using the pico_rgb_keypad and micropython. Is this likley to be added to the current libraries?

Gadgetoid, 2021-01-27:
This is not something we can add into our libraries, since it needs to be added into the Pico MicroPython port to be possible.

There's an open issue about that here - https://github.com/raspberrypi/micropython/issues/5 - but it's not clear if/when it might be tackled.

CircuitPython - however - already supports this, and @sandyjmacdonald has posted some example code for RGB Keypad - https://gist.github.com/sandyjmacdonald/b465377dc11a8c83a8c40d1c9b990a90

Pys3nberg, 2021-01-27:
Thank you, i got @sandyjmacdonald example working with circuitpython. This should be enough to get me going 👍

closed 2021-01-27 by Pys3nberg

TechWilk, 2021-12-24:
For anyone landing here from Google, a decent step-by-step guide with CircuitPython is on the fourm here:

https://forums.pimoroni.com/t/pico-rgb-keypad-base-as-a-usb-hid-device-solved/16224/7