pimoroni/pimoroni-pico #35: Can't build C examples - fails on pico_graphics

artgodwin, 2021-01-28:
I've successfully installed the C build environment and built Raspberry Pi's C++ blinky example.

I've not tried to build the pimoroni examples (I have the pico display) and it proceeds for as while in the final 'make' but fails and  stops with :

[ 14%] Building CXX object libraries/pico_graphics/CMakeFiles/pico_graphics.dir/types.cpp.obj
In file included from /home/adrian/Apps/pico/pimoroni-pico/libraries/pico_graphics/types.cpp:4:0:
/home/adrian/Apps/pico/pimoroni-pico/libraries/pico_graphics/pico_graphics.hpp:61:19: error: enclosing class of constexpr non-static member function 'pimoroni::Pen pimoroni::PicoGraphics::create_pen(uint8_t, uint8_t, uint8_t)' is not a literal type
     constexpr Pen create_pen(uint8_t r, uint8_t g, uint8_t b) {
                   ^
/home/adrian/Apps/pico/pimoroni-pico/libraries/pico_graphics/pico_graphics.hpp:45:9: note: 'pimoroni::PicoGraphics' is not literal because:
   class PicoGraphics {
         ^
/home/adrian/Apps/pico/pimoroni-pico/libraries/pico_graphics/pico_graphics.hpp:45:9: note:   'pimoroni::PicoGraphics' is not an aggregate, does not have a trivial default constructor, and has no constexpr constructor that is not a copy or move constructor
libraries/pico_graphics/CMakeFiles/pico_graphics.dir/build.make:62: recipe for target 'libraries/pico_graphics/CMakeFiles/pico_graphics.dir/types.cpp.obj' failed
make[2]: *** [libraries/pico_graphics/CMakeFiles/pico_graphics.dir/types.cpp.obj] Error 1
CMakeFiles/Makefile2:1664: recipe for target 'libraries/pico_graphics/CMakeFiles/pico_graphics.dir/all' failed
make[1]: *** [libraries/pico_graphics/CMakeFiles/pico_graphics.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Presumably I'm missing something I needed to install but although a long term C programmer I quickly get lost in C++ definitions. Any idea what's going wrong ?

Gadgetoid, 2021-01-29:
Looks like your compiler and ours disagree about what we can get away with in a “constexpr” type function.

Insofar as I can understand any of this... it makes some solid points.

Basically “create_pen” cannot be evaluated constant at compile time because it’s a non-static member of a non trivial non constant class.

Just dropping the “constexpr” from the pico_graphics.hpp may well get you unblocked.

artgodwin, 2021-01-29:
I can try that, but I guess I need to change compilers then.
I'm on debian 9 with gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
What's needed ?

BenHall, 2021-02-16:
I had the same issue on Windows 10 with WSL.

Removing the `constexpr` from `pico_graphics.hpp` allowed it to build

artgodwin, 2021-02-16:
Thanks - I got it to build using the RPi toolset (on an actual Pi) instead
of my default Debian setup. But that's a useful workaround.

automaticdai, 2021-03-27:
Same here (Ubuntu 18.04). Solved the issue by

> Removing the `constexpr` from `pico_graphics.hpp` 

Thanks.