pimoroni/pimoroni-pico #36: pico-audio examples

sedurCode, 2021-01-30:
Hi, 
It appears there is no pico-audio example project. 
Are you in the process of / do you intend to develop one?

Gadgetoid, 2021-03-31:
Hopefully we'll have something to backport from PicoSystem into here as a basic example, but it probably wouldn't be much use to people since there are so many ways you can approach audio- depending on what you want to do!

In the mean time I've raised PR #113 to strip the playground I2S example down to just I2S support and add it to our repository.

I have an audio out board in my possession, and am open to suggestions. What do you want to achieve with audio output?

Note: MicroPython support requires MicroPython first figure out how they want to support I2S, and then it requires support explicitly for the RP2 port, see: https://github.com/pimoroni/pimoroni-pico/issues/96

closed 2021-03-31 by Gadgetoid

sedurCode, 2021-05-24:
Thanks for doing this