pimoroni/pimoroni-pico #37: Instructions don't work for building examples on Windows

conoro, 2021-01-31:
Once I have everything setup as per the main Pico Getting Started doc, I had to use the following commands on Windows to build your examples:

```
cmake -G "NMake Makefiles" ..
nmake

```

CarlSargunar, 2021-02-07:
I'm also finding it really frustrating getting these working on  windows.

Gadgetoid, 2022-01-25:
I think our official party line is "use WSL". We should probably update the documentation to reflect that. 

I've worked on this and other C/C++ projects fairly effectively using WSL1 and VSCode, and Microsoft backing the ol' WSL horse in the first place suggests they don't have much faith in native toolchains either.

closed 2022-06-27 by Gadgetoid

znmeb, 2023-03-10:
Sadly, you cannot reliably access Windows host USB devices from inside WSL. It requires installing a command-line tool, running it manually from an Administrator PowerShell terminal, and it often fails to work. So you can do the *build* in WSL, but if you want to debug using the [Raspberry Pi Debug Probe](https://www.raspberrypi.com/news/raspberry-pi-debug-probe-a-plug-and-play-debug-kit-for-12/) or similar, you need to copy the repository to the Windows filesystem after doing the build.

Now that there's an [official Raspberry Pi Windows toolchain installer](https://www.raspberrypi.com/news/raspberry-pi-pico-windows-installer/), I think Pimoroni should consider re-opening this issue. My guess is Pimoroni (or a user - this is open source) can get "build all" to work in VSCode with the official toolchain long before Microsoft makes host USB devices work reliably inside WSL.

znmeb, 2023-03-11:
I have the builds using VSCode in the official toolchain on Windows 11 mostly working. Everything builds except `motor2040`, `inventor2040w` and `cosmic_unicorn`. Should I open a new issue?

I don't have any of those devices; the only one I have is the audio pack and that's building fine.