pimoroni/pimoroni-pico #41: 'urandom' module always produces the same sequences

birdistheword96, 2021-02-05:
I have three RPI Picos and they always produce the same sequence of numbers when I call random.random() with the Pimoroni version of firmware (v0.0.5):
`
'> import random
'> random.random()
0.7111824

'> random.random()
0.4089468

'> random.random()
0.09752237

'> random.random()
0.8465231
`
The same happens with any of the other 'random' functions, they always produce a replicable sequence.

The standard RPi Micropython port firmware (1.14) from the [Official Website](https://micropython.org/download/rp2-pico/) does not present this issue, if I install that on all three devices every I always get different numbers.

sigmaeo, 2021-02-06:
What you see is typical for a pseudorandom generator.
I think you simply have to seed it yourself with some random value.
Typically with an noisy ADC converter value or with time evolved since start or last keypress or something, that will be not the same each time.

Gadgetoid, 2021-02-08:
I've rebased our MicroPython fork on upstream so the next release should - hopefully - fix this.

Gadgetoid, 2021-02-08:
See: https://github.com/pimoroni/pimoroni-pico/releases/tag/v0.0.6

closed 2021-02-11 by Gadgetoid