pimoroni/pimoroni-pico #45: Any chance there will be CircuitPython libraries for these?

joelburton, 2021-02-07:
I'd love to add an IR receiver to my Unicorn project, but the IR decoding libraries I use are for CircuitPython (and there doesn't look like anything similar for MicroPython).

Does anyone have any ideas on where to start with porting this to CircuitPython?

Thanks!

Gadgetoid, 2021-02-08:
Figuring out where to start is on my TODO list. I'm a fan of CircuitPython but we can't just bake our libraries in and ship our own version like we're doing with MicroPython.  The various low-level drivers need merged upstream and then Python wrapped for them created in the same style (using Adafruit's CookieCutter template) as other CircuitPython libraries and PR'd into the Community Bundle - https://github.com/adafruit/CircuitPython_Community_Bundle

harveysandiego, 2021-03-10:
I am keen to get these libraries ported to CircuitPython as well, I've got Python & C/C++ knowledge...  I can just about understand Make & CMake files and have had a cursory glance at how everything is built.

In terms of merging the low-level drivers upstream, what needs to happen there?

As for the CookieCutter stuff, I've had a look and that doesn't seem too complex.

Gadgetoid, 2021-03-10:
Specifically we would need to (and this is very off the top of my head so it may be wrong/incomplete):

1. [ ] Verify there is upstream support for the display parts we're using on Pico Display Pack and Pico Explorer Base
2. [ ] For Pico Unicorn we would need to investigate what's involved with up-streaming the PIO driver required to make it work
3. [ ] We have a loose working list of our Breakout Garden boards that might already be supported by CircuitPython drivers, and ones that need a driver. Our whole library of breakouts still needs testing and any compatibility fixes PR'd against the appropriate library.

The TODO list for breakouts exists in this Wiki - https://github.com/pimoroni/BreakoutGarden_CircuitPython/wiki/Breakout-TODO - which is now a little out-of-date.

Functionality such as buttons and general IO stuff found on Explorer can just be part of CircuitPython examples that show the idiomatic way to toggle or read a pin, plus some documentation detailing which pin is which.

Blist, 2023-10-14:
Hello,

I recently purchased a Pico Unicorn pack along with a Pico Audio pack. I noticed that the audio pack requires CircuitPython while the Pico Unicorn pack requires MicroPython.

Is there a way for both to work with the same library? I'm interested in creating a MIDI visualizer and was drawn to the Pico Piromoni pack. I love the concept of using it for prototyping, but it's unfortunate if all modules can't operate with the same version of Python.

Gadgetoid, 2024-06-04:
Been a long time in the making, but thanks to @thirdr we have some i2s audio examples available. eg: https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/pico_audio

closed 2024-06-04 by Gadgetoid