pimoroni/pimoroni-pico #51: USB Device Descriptor Request Failing

crispythegoat, 2021-02-09:
Using REPL on Windows with the Pimoroni 0.6 Alpha Micropython firmware.

Windows doesn't recognise the USB serial device and throws the following error "A request for the USB device descriptor failed."

Mass storage mode works correctly.

_Using the latest Micropython firmware from Raspberry Pi themselves works correctly and does not result in the above issue._

crispythegoat, 2021-02-09:
Please note this issue was somewhat down to my own naivety.

I had previously uploaded a .py file named main.py (was testing to ensure it remained persistent when not connected to a machine), however there was an error in the code which I presume cause the device to panic upon loading the firmware when booting.

Although it had booted and connected just fine with Raspberry Pi's own Micropython firmware file. So perhaps this still needs to be looked at despite it being somewhat self inflicted?

Gadgetoid, 2021-02-11:
Version 0.6 should be pretty much identical to Raspberry Pi's own release, or is at least built from a pretty fresh fork of upstream MicroPython. I'm not familiar enough with MicroPython to know why this might be happening but I'm willing to roll with user error. Let us know if it crops up again!

closed 2021-03-31 by Gadgetoid