pimoroni/pimoroni-pico #52: picoexplorer.get_adc(x) hangs

lowfatcode, 2021-02-10:
When attempting to read an ADC channel from Micro Python the Pico hangs completely.

lesp, 2021-02-10:
Thonny shell becomes unresponsive, and I have to power off the Pico to regain shell.

[image]

UnfinishedStuff, 2021-02-10:
After a bit of poking, this seems to happen if you don't call `picoexplorer.init()` first.  You need to pass `init()` a bytearray, but if you're not using the screen it can be an empty array.

Presumably this is because `init` also initialises the ADCs.

ZodiusInfuser, 2021-02-11:
Thanks for pointing this out @lesp! As @UnfinishedStuff mentions, the cause of the problem is that you're not calling `picoexplorer.init(<bytearray>)` before you attempt to read the ADC, which means an instance of PicoExplorer does not exist at that point.

This is something we should be handling gracefully, so what I'll do is have each function fire off a RuntimeException in this case that will instruct the user that init needs to be called, like:

[image]

Quick question though. How should I do the wording for that error message?

ZodiusInfuser, 2021-02-11:
#54

closed 2021-02-11 by ZodiusInfuser