pimoroni/pimoroni-pico #57: pico_display.text() renders mixed case as UPPER CASE

rupertreynolds, 2021-02-14:
Example prints "Hello" as "HELLO" using v0.0.6a

Please can you tell me, Is this likely to change in future?

Roops

rupertreynolds, 2021-02-14:
disp.set_pen(192,0,0)
disp.text("Hello",100,50,100,4)
disp.update()

Pinkyatacme, 2021-02-15:
The default 6x6 font has the same glyphs for upper and lower case. I submitted a PR (#55) of the updates I made to support different height fonts (although it doesn't include my 6x8 font that led to me making the update (I have a custom character replacing the backtick, but I guess I should do a formal 6x8 sample as a PR!)

Pinkyatacme, 2021-02-15:
FYI - PR #55 has been updated to include a 6x8 font (rename to font_data.cpp to use it, after renaming the current one out of the way). In use, you may need to adjust any y scale positioning to allow room for the taller text.

Gadgetoid, 2021-03-30:
This should be merged now, however fonts are attached as libraries and used as per this demo:

https://github.com/pimoroni/pimoroni-pico/blob/main/examples/pico_explorer/text_demo.cpp#L33

Not yet supported in MicroPython!

Gadgetoid, 2022-06-27:
Pico Graphics has been completely overhauled, `display.set_font("bitmap8")` supports upper/lower case and some accented chars now. Vector fonts support upper/lower but ASCII only.