pimoroni/pimoroni-pico #58: pico_display

KeithWatterson, 2021-02-14:
This example code locks up my the pico device when c ~ 35 to 40. If I rem to the display.text line it works fine.
```python
import machine
import picodisplay as display
import time

led_onboard = machine.Pin( 25, machine.Pin.OUT )

width = display.get_width()
height = display.get_height()
display_buffer = bytearray( width * height * 2 )
display.init( display_buffer )
display.set_backlight(1.0)

count = 0
while True:
    led_onboard.toggle()
    
    count += 1
    t = "{:.1f}".format( count / 10 )
    print( t )
    
    display.set_pen( 0, 0, 0 )
    display.clear()  
    display.set_pen( 255, 255, 255 )
    display.text( t, 1, 40, 240, 4 )
    display.update()
    time.sleep( 1.0 ) 
```

helgibbons, 2021-02-14:
I'm getting this too - seems to be an bug in v0.0.6 / v0.0.7 of the MP firmware as v0.0.5 iterates as it should without falling over

SimUKdev, 2021-02-15:
I'm getting a similar issue with the Pico Explorer. It seems to be when trying to include a variable as the "text" part rather than specifying an actual string of text in quotes, such as:

`explorer.text("words here", 20, 95, 100)`  _works as expected_

`explorer.text(adc0, 20, 95, 100)`  _causes a lockup/requires a Pico reboot_

It doesn't seem to crash with any error, just makes Thonny (on my RPi 4) lock up, and clicking the Stop icon just returns a horizontal line each time. Only a disconnect and reconnect of the Pico's USB cable appears to allow it to continue working. 

One possible workaround I've found, seems to be to have the value stored as a string, such as _stringadc0_ below:

```
adc0 = int(explorer.get_adc(0) * 100) # get the adc0 reading
stringadc0 = str(adc0) # create a string version for including in the explorer.text
print(adc0) # print to shell to confirm the value
print(stringadc0) # print to shell to confirm the value
explorer.text(stringadc0, 20, 120, 100, 5) # this will display the adc0 value on the Pico Explorer screen. Using the value not converted to a string does not and makes the Pico/Thonny lockup
```

However the lack of error message and lockup makes me feel that whilst converting to string "works", it's probably not really solving the underlying issue.

Gadgetoid, 2021-03-30:
Attempting to display a `float` or `int` with `text()` seems to crash the device pretty hard- this will be a bug in the Python bindings which no doubt fail to verify that the input object is a string, or fail to handle non-string types.

As for the code posted above... I can reduce the sleep time to speed things up and run this apparently indefinitely. Right now I'm on 1400 iterations. This is with v0.0.9, though, so it's possible a bug was fixed somewhere.

closed 2021-03-30 by Gadgetoid