pimoroni/pimoroni-pico #59: Suggestion: pico_display API

sumitkm, 2021-02-14:
Since the display needs a call to `init()` to be usable, can we consider setting `set_pen(0,0,0)` and calling `clear()`, internally, the first time `init` is called?

Not really a big deal. 

I am trying to figure out what it takes to make a pull request, but a complete n00b at Micropython build environment. Still trying to wrap my head around how to test my changes :-)

Pinkyatacme, 2021-02-23:
Personally, I would suggest that a overloaded call to init, with RGB values for the background would be preferable. If you look at the display demo code, that uses a rather nasty brownish colour for the background, so a black background, while likely to be the most common, is certainly not an automatic choice in all cases.

helgibbons, 2022-07-11:
Closing as relates to old pico_display API.

New shininess here!: https://github.com/pimoroni/pimoroni-pico/blob/main/libraries/pico_graphics/README.md

closed 2022-07-11 by helgibbons