pimoroni/pimoroni-pico #62: Pico Explorer MicroPython library GPIO GP0 to GP7 functions?

SimUKdev, 2021-02-18:
As far as I can tell, there doesn't appear to be any functions in the Explorer MicroPython library to control GP0-GP7, beyond the function to set one(?) of them as the audio pin's signal source. For reference, from Lines 5 to 14, the following functions have been declared for users to make use of:

> _L5-14_ https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules/pico_explorer/pico_explorer.h#L5

```
// Declare the functions we'll make available in Python
extern mp_obj_t picoexplorer_init(mp_obj_t buf_obj);
extern mp_obj_t picoexplorer_get_width();
extern mp_obj_t picoexplorer_get_height();
extern mp_obj_t picoexplorer_update();
extern mp_obj_t picoexplorer_is_pressed(mp_obj_t button_obj);
extern mp_obj_t picoexplorer_get_adc(mp_obj_t channel_obj);
extern mp_obj_t picoexplorer_set_motor(mp_uint_t n_args, const mp_obj_t *args);
extern mp_obj_t picoexplorer_set_audio_pin(mp_obj_t pin_obj);
extern mp_obj_t picoexplorer_set_tone(mp_uint_t n_args, const mp_obj_t *args);
```

In comparison, the Explorer Hat on the full size Pi appears to have functions like `explorerhat.input.one` [input](https://github.com/pimoroni/explorer-hat/blob/master/documentation/Function-reference.md#input) and [output](https://github.com/pimoroni/explorer-hat/blob/master/documentation/Function-reference.md#output) to be able to interact with the input and output pins on that board.

Is Pico Explorer meant to have built in functions for the GP0-GP7 pins, or is it expected that users will do `from machine import Pin` and use that way to interact with them, rather than as part of the Pico Explorer's MicroPython library?

ZodiusInfuser, 2021-02-20:
I believe it was expected that you would use `from machine import Pin` as you say, as that gives you full control of what those pins can do, which may not be possible with whatever we expose.

That being said, if you think we should still add a function like you suggest then let us know and we can look into it.

helgibbons, 2021-03-30:
This is hopefully made a little less confusing with #91 :)

closed 2021-03-30 by helgibbons