pimoroni/pimoroni-pico #63: Let your little LCD shine. "Balls stay in court".

MarkEmery, 2021-02-18:
I'm really liking the pico display. The bouncing balls demo irked me a little, as the ball bounced on a line that's off the frame. I went to do a Pull Request, couldn't work out how to do it for the examples. Here's a tweaked version that takes the ball size into account so each bounces as it touches the edge of the screen. It doesn't give the viewer the false impression that the screen is somehow cropped. It also sets the speed of the ball in proportion to its size. Smaller = faster.
If you like it, feel free to use it.
[new-balls.py.txt](https://github.com/pimoroni/pimoroni-pico/files/6006430/new-balls.py.txt)

Gadgetoid, 2021-02-23:
Love it! Thank you.

*waves git magic wand*

https://github.com/pimoroni/pimoroni-pico/pull/69

closed 2021-02-23 by Gadgetoid

dsssssssss9, 2021-02-24:
Hi mark 
really loving what you've done to this demo - thank you so much as it was bugging the heck out of me having the balls seemingly bounce off of something just off screen!!

Appreciate it very much!

MarkEmery, 2021-02-24:
> Mark Emery Limited mark@markemerylimited.co.uk Tel: 077 6997 3757 Incorporated in England and Wales number 7065343. Registered Office: C/o Calder & Co, St Vincent House, 30 Orange St, London WC2H 7HF. VAT Registration Number 109756787

You’re welcome! Loving your products. You’re now my go-to shop over all others.

Regards,
Mark