pimoroni/pimoroni-pico #72: Picodisplay and Threads?

callimero, 2021-02-24:
It would be nice to use threads to double speed or put drawing into another core right?

I did some test and "it sorta works". There is garbage at the screens last few lines as soon as I start the second thread. 
X Starts the Thread, A ends it, B ends the main

But maybe (most probably) I just did something wrong with my barefoot aproach?

`
import random
import picodisplay as display 
import time, _thread, machine

width = display.get_width()
height = display.get_height()

display_buffer = bytearray(width * height * 2)  # 2-bytes per pixel (RGB565)
display.init(display_buffer)

display.set_backlight(1.0)

display.set_pen(255, 255, 255)        
display.clear()

radius=5

def task(s, radius):
    print(width,height,dir(display))
    while not(display.is_pressed(display.BUTTON_A)):
        display.set_pen(0, 250, 0)        
        display.rectangle(random.randint(0,width-1), random.randint(0,height-1), radius, radius)
        time.sleep(s)
        print("-")

#_thread.start_new_thread(task, (0.2, 10))

while not(display.is_pressed(display.BUTTON_B)):
    display.update()
    print("+")
    display.set_pen(255, 0, 0)        
    display.rectangle(random.randint(0,width-1), random.randint(0,height-1), radius, radius)
    time.sleep(0.2)
    if display.is_pressed(display.BUTTON_X):
        _thread.start_new_thread(task, (0.2, 10))

`

callimero, 2021-02-24:
Adding some lines to the height buffer clears the display at the bottom... hacky :)

Gadgetoid, 2021-03-31:
I think this is related to #86 and #89