pimoroni/pimoroni-pico #73: Custom firmware?!

callimero, 2021-02-24:
I now tried hours to compile a custom firmware.

I want a micropython with picodisplay and ulab (similar to numpy).

I can make a new firmware from the stock rp2 tree, it makes a firmware.uf2 with no problems. 

But I cant get an idea how to include own modules, I tried with stuff like 

`make  USER_C_MODULES="../../../pimoroni-pico/micropython/modules" CFLAGS_EXTRA="-DMODULE_PICODISPLAY_ENABLED=1" all`

in quite a lot combinations... Any hints welcome I guess I miss something simple...

Gadgetoid, 2021-02-25:
The rp2 port of MicroPython uses CMake instead of makefiles so it doesn't (yet) have user C module support.

See: https://github.com/micropython/micropython/pull/6960

We maintain our own fork of MicroPython with these changes so we can build our custom firmware - https://github.com/pimoroni/micropython/tree/continuous-integration

callimero, 2021-02-25:
Ah thanks! So no wonder :)
I will have a look into.

callimero, 2021-02-26:
So two days later... I somehow can now make the pimoroni branch and include the picodisplay module!

I combined several  tutorials, not sure about the way which lead to success, so much is in flow. None of the tutorials worked from a--z.

I also tried to add ulab (numpy similar) but this seems not to play with Pico .

I really wish we could have a webservice one day which is curated an can be build a firmware of your liking....

borgsoftwaresystems, 2021-03-24:
I want to do similar. I've forked the Pimoroni library and changed it for better font support. I want to make this available from MicroPython but I have no idea how to role my changes into a custom MicroPython firmware. To make matters harder I need to do this from a Windows 10 build machine rather than Raspberry Pi. I asked Pimoroni support but they couldn't help me.

Gadgetoid, 2021-03-24:
The build process for custom modules in the rp2 port has also changed as of yesterday since I’m taking onboard suggestions in order to get it fit for merge into MicroPython.

GitHub actions are set up to already build MicroPython from source any time I publish a release so perhaps the easiest way to distribute a custom version is just to publish a release on your fork... bit of a rigamarole if you’re wanting to build and test locally with any kind of speed though!

A website to build custom versions based on a tick box list of available modules is something I’m really keen to investigate.

callimero, 2021-03-24:
Yea, the "pimos" are quite busy, but that all makes it a bit hard to help for the community.

callimero, 2021-03-24:
> The build process for custom modules in the rp2 port has also changed as of yesterday since I’m taking onboard suggestions in order to get it fit for merge into MicroPython.
>
> GitHub actions are set [...]

In the early day I did see such sites quite often for ESP firmwares.

closed 2021-03-24 by callimero