pimoroni/pimoroni-pico #77: standalone Libraries/packagesfur use with thonny

CAProjects, 2021-02-27:
I am going to be blunt because someone needs to.

Custom with your own packages is a way to fragment a community especially one that is just beginning. Why do a custom firmware instead of making the libraries/packages available though pypi so that people can use your hardware with others?

If the Pico keeps going down the route of everyone one who makes a product for the Pico has their own custom firmware and no packages then its going to be a terrible future for the Pico

Please put the packages/libs/whatever on pypi for those of us that don't want/know how to build what is needed. Your current way is NOT user friendly and is 1 thing holding me back from purchasing your products that I really want

Gadgetoid, 2021-02-27:
I understand where you're coming from, but we *need* to write C++ libraries for the Pico SDK and don't have the people power to write separate libraries for:

1. Pico SDK C++
2. CPython
3. MicroPython
4. CircuitPython

Insofar as I understand, it's not possible to ship our C libraries on PyPi for MicroPython anyway, which makes anything requiring custom-code a non-starter unless we put the extra effort in to upstream it. And that's no simple process.

I don't see how shipping an identical copy of MicroPython with all of our libraries baked in *isn't* "user-friendly". Though. What are you losing? Copying libraries to the Pico is a pain, so I'd venture it's *more* user-friendly.

I've put some considerable work into making this approach viable by fostering and attempting to upstream the code we rely upon so our MicroPython build can be pretty much 1:1 with the "official" release - https://github.com/micropython/micropython/pull/6960

Gadgetoid, 2022-03-30:
Closing for now because I'm still keenly aware it would be *nice* to for there to be more generic MicroPython libraries, but it's not going to happen imminently.

closed 2022-03-30 by Gadgetoid