pimoroni/pimoroni-pico #80: Suggestion: Screen orientation setting

Galbi3000, 2021-03-02:
The picodisplay seems to be fixed orientation so that the "Pico Display" logo on the silkscreen is on the left. But the way I need to use it due to the USB cable I need the logo on the right so now everything is upside-down.

On the Arduino display libraries, either by Adafruit or other compatible libraries, there is a function to rotate the display so any of the edges are the top edge so you can have it 0,90, 180 or 270 degrees rotated.

Any chance a similar function could be added to the display library and module here?

Pinkyatacme, 2021-03-03:
A minor adjustment to the st7789 driver code inverts the display (on line 59 change "\x70" to "\xB0". As there's only a minor adjustment needed to flip the display in both landscape and portrait formats (inverting bits 6 & 7 of the MADCTL value), it should ab fairly simple to add an additional parameter to the display.

hamid-elaosta, 2021-03-04:
Check out my PR from about a week ago if you want a quick way to change orientation. https://github.com/pimoroni/pimoroni-pico/pull/75

Pinkyatacme, 2021-03-04:
I have a local version that implements this (thanks for the pointers Hamid), but it's not exactly "correct". On looking at the code, doing it properly is a little more complex, as the 'command' function expects a string (that is then converted into a uint8!), so a simple XOR of the MADCTL value isn't quite as simple as I had hoped.

hamid-elaosta, 2021-03-04:
This library from Adafruit has an orientation function, perhaps it could be used for implementation pointers?

https://github.com/adafruit/Adafruit-ST7735-Library/blob/master/Adafruit_ST77xx.cpp

Pinkyatacme, 2021-03-06:
Just added PR #85 to implement a flip command - a call of pico_display,flip() [assuming that's your display object name!] should rotate the screen 180. I would have preferred to use the RDDMADCTL call to fetch the current data, but struggled to get that working (it seems to need a dummy spi read to get to the data, and I can't seem to figure that one out).

Galbi3000, 2021-03-07:
> Just added PR #85 to implement a flip command - a call of pico_display,flip() [assuming that's your display object name!] should rotate the screen 180. I would have preferred to use the RDDMADCTL [...]

That would solve my issue perfectly. I don't have the means to compile the source so I will have to wait for it to be added to the main branch and the firmware image file is updated.

Pinkyatacme, 2021-03-08:
OK - I haven't actually taken a look at the Python side of this yet, but will have a look. Hopefully it's not too complex, as the Python code I've looked at seems to be made up of calls to C for most of the modules.

Pinkyatacme, 2021-03-08:
@Galbi3000 - I've got an updated firmware image if you want to try it: http://www.acmelabs.co.uk/wp-content/uploads/2021/03/firmware.zip

Galbi3000, 2021-03-08:
> OK - I haven't actually taken a look at the Python side of this yet, but will have a look. Hopefully it's not too complex, as the Python code I've looked at seems to be made up of calls to C for most of the modules.

> @Galbi3000 - I've got an updated firmware image if you want to try it: http://www.acmelabs.co.uk/wp-content/uploads/2021/03/firmware.zip

I assume from the first quote there that you have only implemented the flip in the C code? At this time I have not set up a C environment for the Pico, I've just been playing around with it in Python.

Pinkyatacme, 2021-03-08:
The firmware.uf2 in the zip file is an updated MicroPython image, built on the core Pimoroni one, but with the addition of the flip function. I prefer to work in C++ on embedded stuff, but built (and tested) this to make sure I understand how to update the MicroPython image.

Galbi3000, 2021-03-08:
I'll give it a go then :)

Gadgetoid, 2022-03-31:
This is a TODO item as part of #327 which should overhaul how we handle displays to be a little more generic and flexible.

closed 2022-03-31 by Gadgetoid