pimoroni/pimoroni-pico #83: PicoDisplay not working

sdalogic, 2021-03-05:
I have just taken delivery of a PicoDisplay module and am trying to use this with a new Pico. I have installed the Pimoroni customer firmware and have tested with all four of your examples from GitHub, none of these work.
The RGB led at the bottom does light, but nothing on the display and no response when the buttons are pressed in the relevant example. I know the code is running in the button example as I have added some 'print' statements, but it never responds to any button press.
Could this be a faulty device?
Help please.

Gadgetoid, 2021-03-05:
Could you post photos of your setup & header soldering on the Pico? These are new and shiny, so I've very little troubleshooting experience but my first guess is always some kind of connection issue.

sdalogic, 2021-03-06:
Hi Philip,

 

I really don’t think this is a connection issue. I have attached two images of the Pico with the Display device attached, also another image of a Pico without any header attached so you can see better the quality of my soldering. The soldering has been done very carefully and checked under a magnifying lamp I use, all joints were perfect on all three of the Picos I have been using.

Before attaching the Display header, I checked some simple code on the Pico involved and all was fine. I don’t believe it is a Pico problem, I suspect the Display device.

 

Peter

 

Peter Byrne BSc(Hons), FCA

Director

 

SDA Logic Limited

United Kingdom

 

Telephone: 01489 894 180    l    Mobile: 07860 487 956    l    EMail:  <mailto:pbyrne@sdalogic.com> pbyrne@sdalogic.com

 

This e-mail has been scanned for all viruses by Claranet. The
service is powered by MessageLabs. For more information on a proactive
anti-virus service working around the clock, around the globe, visit:
http://www.claranet.co.uk

This e-mail has been scanned for all viruses by Claranet. The
service is powered by MessageLabs. For more information on a proactive
anti-virus service working around the clock, around the globe, visit:
http://www.claranet.co.uk