pimoroni/pimoroni-pico #90: Using pimoroni sdk outside of examples tree

citizenfish, 2021-03-10:
If I want to code a programme based on your examples but outside of the pimoroni source tree, how would I include the libraries?

eg:-

-pimoro-pico
-pico-sdk
-my_project_directory

I think this would be useful for (ancient) developers like me not overly familiar with cmake

citizenfish, 2021-03-11:
I have solved this by adding:-

```
include(../pimoroni-pico/libraries/pico_explorer/pico_explorer.cmake)
```

To CmakeLists.txt

Gadgetoid, 2022-01-26:
The Pico Boilerplate is a good starting point for using external libraries in your own project - https://github.com/pimoroni/pico-boilerplate

It also has the necessary GitHub action setup and so forth to continuously build/test your code.

closed 2022-01-26 by Gadgetoid