pimoroni/pimoroni-pico #94: Pico Explorer Base schematic

bgolab, 2021-03-17:
Hi,
Just received the board.
I cannot find the schematic.

Could please let me know the link to the schematic?

ZodiusInfuser, 2021-03-18:
Hi. We currently do not have schematics available for the PicoExplorer or our other Pico products. We do however have pinout diagrams on each product page. Here is the one for PicoExplorer:
[image]

closed 2021-03-24 by ZodiusInfuser

Jedzia, 2021-07-29:
So you "do not have schematics available for our pico products" and your primary customer base are makers.
Are you some kind of Troll-Company?

Joking aside, this is embarrassing. Do your work properly and publish the necessary documents. If I had known that beforehand, I would not have bought anything from you. Such behaviour is more likely to be seen in amateurs and botchers. 

No circuit diagrams at a company that develops and sells electronic components. A really bad joke!

Jedzia, 2021-07-29:
Oh fuck, via https://forums.pimoroni.com/t/explorer-hat-pro-schematics/3487/4

> gadgetoid
> Pimoroni Crew
> deonm
> Dec '16 2016
>
>> Deon _[a user asking, Jedzia]_; 
>>     I cannot think of a good reason why you do not publish the schematics.All the pertinent information is on the links you posted above.
> 
> It would take time to tidy them up, annotate them properly and consistently and release them. We’re working on it, slowly, but it’s not a priority. Our Explorer HAT schematic still has “Activity HAT ???” in the top left corner!

At the rest of the world, engineers have a plan(a readable one!) before they build or release something, at least those with a proper education.
You **ARE** amateurs and botchers!

Let me help you advertise your unique selling proposition (= no circuit diagrams). Wherever I can and completely free of charge.
That's a good joke;)

And completely free of charge (well, lawyers) I can find out whether this is unfair advertising on your part, to conceal such an important point on your shop website.
It is expected that proper documentation (circuit diagram!) Is available, especially for hats, extensions, additional boards or the like. Such as Adafruit, seeedstudio, Arduino, etc. show that by good example.

helgibbons, 2021-07-29:
The Pico Explorer schematic is linked from the shop page: https://cdn.shopify.com/s/files/1/0174/1800/files/pico_explorer_schematic.pdf?v=1619077703

Gadgetoid, 2021-07-29:
If you’d like to post abusive comments about us that’s fine. But I’d highly recommend social media or your own personal blog. I expect civil conduct on GitHub issues since these are public and our audience reaches various age groups.

If schematics are important to you a simple “schematics are important to me and I’d like you to consider making the time to prioritise them” would have sufficed. A stream of abuse is quite the opposite of motivating.

Jedzia, 2021-08-02:
Sorry if my "abusive" critique hurt your feelings, or industry standards hurt your feelings.
My point is that money is a better way to push you to a more professional way of acting.

I'm not interested in your crocodile tears, nor I am your motivation trainer. I want schematics!

(Imagine a crowd screaming "MAOAM, MAOAM, MAOAM". I took some of your last post as a joke, especially that with the social media[Github: Am I a joke to you? lol]. This is my joke to be even. Further imagine politicians or companies telling people WHERE and HOW to demonstrate, tell their critique or speak out loud. Are you sure you want THAT as a MEME surrounding you on, e.g. Twitter? Sorry but I am laughing bloody tears, now)