pimoroni/pimoroni-pico #96: Pico Audio python api

joseph-abell, 2021-03-19:
Hi

Is there any place for me to watch for updates regarding the pico audio python api? Sorry if I've missed it, but I couldn't see anything other than it is planned.

I'm making a mini theremin and I'd very much prefer to work in python rather than c, and this is the only part remaining.

Gadgetoid, 2021-03-30:
I suspect Python support for audio might be some time off- it's still early days for Pico's MicroPython port and there's still no precedent set - as far as I can tell - for analog audio or digital audio (I2S). Though the latter seems to be a work in progress for esp32 at least - https://github.com/micropython/micropython/pull/4471

I'm not sure if we'd be able to meaningfully support anything resembling audio in the meantime. I'm at least familiar with the concepts, but not how they might interact with MicroPython.

joseph-abell, 2021-07-23:
It looks like https://github.com/micropython/micropython/pull/7183 has given micropython an I2S api. Is this relevant / useful?

Gadgetoid, 2022-01-25:
i2s support has been added to the RP2 port as of MicroPython 1.18 - we're going to have to test this and put together some kind of example, but it will at least form the groundwork for an audio API: https://github.com/micropython/micropython/releases/tag/v1.18

ned-pcs, 2023-05-10:
There is a problem with the I2S handling in micropython if you're using the HUB75 support from pimoroni: the HUB75 handler uses both DMA complete interrupts, which means that the I2S won't be able to use the one it needs.

Gadgetoid, 2024-03-11:
> There is a problem with the I2S handling in micropython if you're using the HUB75 support from pimoroni: the HUB75 handler uses both DMA complete interrupts, which means that the I2S won't be able to use the one it needs.

Near as I could tell, the second DMA complete interrupt wasn't actually used for anything. I dropped it and tidied up the Hub75 code in this commit: https://github.com/pimoroni/pimoroni-pico/commit/b0d53dadb38ada44d8911f14c6f60573b41d7278

MicroPython now uses shared handlers across DMA/i2s, too.

We're still missing examples of actually using i2s audio, though there are some here: https://github.com/miketeachman/micropython-i2s-examples

closed 2024-03-11 by Gadgetoid