pimoroni/pimoroni-pico #97: Pico doesnt run properly once main.py is saved on the board

davek61, 2021-03-19:
Hello,
I am using a pico display and it worked ok'ish  until I saved my program on to the pico. 

After this I have been unable to run programs from RAM.  It keeps disconnecting or coming up with odd compile errors.

It was noticeably quite unreliable with regards to the USB serial connection even before this problem was observed.  I often had to disconnect and reconnect before I could get the code to run in flash.

I an using the latest pimoroni python firmware downloaded today

Gadgetoid, 2021-03-30:
Presumably this is because `main.py` is run at boot and trounces all the memory before you can get around to using it.

I've started avoiding saving as `main.py` for this reason, and it's still flaky :roll_eyes: - I don't think it helps that Thonny distinguishes between "Run" and "Stop -> Run" and if I fail to explicitly *stop* before each successive run of my code, it'll fall apart.

We're pretty much toe to toe with upstream MicroPython, though, so I'm guessing this is just a fact of life with MicroPython on Pico (exacerbated by the fact our products have big displays and allocate big display buffers) and I honestly don't think there's much we can do about it.

helgibbons, 2021-03-30:
@davek61 If your `main.py` is broken to the extent that it locks up your Pico and Thonny can't communicate with it any more to _stop_ it, you can copy [this file](https://www.raspberrypi.org/documentation/pico/getting-started/static/6f6f31460c258138bd33cc96ddd76b91/flash_nuke.uf2) to it to clear the flash memory and start again. You'll need to copy the MicroPython firmware across again afterwards.

Gadgetoid, 2024-03-11:
Closing as stale/complete.