pimoroni/pimoroni-pico #99: update() is crashing the Pico with a Pico Display attached

nickwebcouk, 2021-03-21:
This is probably the worst issue I've had to raise as I have no idea what's happening.

I have 300-odd lines of hacked together python code that almost works - but every time I call display.update() the whole pico crashes. Theres no update, there's no screen change, there's no output on the console - it just freezes.

I'm certain a variable hasn't been set correctly but I can't see any issues. Theres no errors - the board just crashes. Because its Micropython I Cant figure out how to enable any advanced logging or debugging so I've included my code here.

The actual code is in a created function and receives the data through that function, but is reproducible using the variables defined here.

Sorry for the amount of print statements - but it was me trying to get to the bottom of what's happening. All variables are printing correctly, so I've no idea what's happening.

I've tried on both Thonny and Visual Studio Code. The code is being run as its typed, so not saved as main.py yet.

I have the Pico Display hooked up to the board directly, with 4 pins that are not in use repurposed to connect a Neo GPS Module.

`import time
import picodisplay as display 

width = 240
height = 135
display_buffer = bytearray(width * height * 2)  # 2-bytes per pixel (RGB565)
display.init(display_buffer)
display.set_backlight(1.0)

counter = 0

previous = 0
duration = 500
led = 0
currentTime = 0

red = 0
green = 0
blue = 255

FdateGPSStr = "21/03/21"
FtimeGPSStr = "22:23:36"
FfixTypeStrA = "No"
FfixTypeStrB = "Warn"
FsatUseInt = 0
FspeedMPHFlt = 0.0

#def updateAttachedDisplay(FdateGPSStr, FtimeGPSStr, FfixTypeStrA, FfixTypeStrB, FsatUseInt, FspeedMPHFlt):  

print(FdateGPSStr + FtimeGPSStr + FfixTypeStrA + FfixTypeStrB + str(FsatUseInt) + str(FspeedMPHFlt))
# Requires the following variables to be passed

# FdateGPSStr     (Passed straight to display)
#   - 21/02/21
# FtimeGPSStr     (Passed straight to display)
#   - 04:58:20
# FfixTypeStrA    Checked and background colour changed on value
#   if No - Red background
#   if 2D - Yellow background
#   if 3D - Green background
# FfixTypeStrB    Checked and background colour changed on value
#   if Warn - Red background
#   if Valid - Green background
# FsatUse         (Passed straight to display)
# TODO:
# FspeedMPH       Checked and limited to prevent spurious readings
#   Format first - no more than 3 decimal places (No point - precision is to 3 decimal places)

# LED Setup:
# if FfixTypeStrA is No
if FfixTypeStrA == "No":
    FfixTypeStrAbg = display.create_pen(255, 0, 0)
# AND
# if FfixTypeStrB is warn = RED
    if FfixTypeStrB == "Warn":
        red = 255
        green = 0
        blue = 0
# OR
# if FfixTypeStrB is Valid = YELLOW
    if FfixTypeStrB == "Valid":
        red = 255
        green = 255
        blue = 0

# if FfixTypeStrA is 2D
if FfixTypeStrA == "2D":
    FfixTypeStrAbg = display.create_pen(255, 255, 0)
# AND
# if FfixTypeStrB is warn = YELLOW
    if FfixTypeStrB == "Warn":
        red = 255
        green = 255
        blue = 0
# OR
# if FfixTypeStrB is Valid = YELLOW
    if FfixTypeStrB == "Valid":
        red = 255
        green = 255
        blue = 0

# if FfixTypeStrA is 3D = Green
if FfixTypeStrA == "3D":
    FfixTypeStrAbg = display.create_pen(0, 255, 0)
# AND
# if FfixTypeStrB is warn = YELLOW
    if FfixTypeStrB == "Warn":
        red = 255
        green = 255
        blue = 0
# OR
# if FfixTypeStrB is Valid = GREEN
    if FfixTypeStrB == "Valid":
        red = 0
        green = 255
        blue = 0
print("DEBUG INV 1")
if FfixTypeStrB == "Warn":
    FfixTypeStrBbg = display.create_pen(255, 0, 0)
elif FfixTypeStrB == "Valid":
    FfixTypeStrBbg = display.create_pen(0, 255, 0)
print("DEBUG INV 2")
if FsatUseInt < 4:
    FsatUsebg = display.create_pen(255, 0, 0)
elif FsatUseInt >= 4 and FsatUseInt <= 6:
    FsatUsebg = display.create_pen(255, 255, 0)
elif FsatUseInt > 7:
    FsatUsebg = display.create_pen(0, 255, 0)
else:
    FsatUsebg = display.create_pen(0, 0, 255)

print("DEBUG INV 3")
if FsatUseInt <= 9:
    FsatUseDisplay = "0" + str(FsatUseInt)
elif FsatUseInt > 9:
    FsatUseDisplay = FsatUseInt
else:
    FsatUseDisplay = "Er"

print("DEBUG INV 4")
#print(FspeedMPH)
#print("test" + str(round(FspeedMPH,2)))

FspeedMPHStr = round(FspeedMPHFlt, 3)
print("DEBUG INV 5")

currentTime = time.ticks_ms()

if (currentTime - previous) >= duration:
    if led == 0:
        display.set_led(red, green, blue)
        previous = currentTime
        led += 1
    else:
        display.set_led(0, 0, 0)
        previous = currentTime
        led = 0    
print("DEBUG INV 6")

print("Debug Display Line Values: ")
# TOP LEFT, DATE AND TIME
display.set_pen(0, 0, 0)
display.clear()
display.set_pen(255, 255, 255)
display.text(FdateGPSStr, 0, 0, 240, 4)
display.text(FtimeGPSStr, 0, 30, 240, 4)
print(FdateGPSStr)
print(FtimeGPSStr)

# TOP RIGHT, PART 1, 2D/3D FIX AND BACKGROUND COLOUR
display.set_pen(FfixTypeStrAbg)
display.rectangle(180, 0, 90, 30)
display.set_pen(0, 0, 0)
display.text(FfixTypeStrA, 190, 0, 240, 4)
print(FfixTypeStrA)

# TOP RIGHT, PART 2, SAT NUMBER AND BACKGROUND COLOUR
display.set_pen(FsatUsebg)
display.rectangle(180, 30, 90, 30)
display.set_pen(0, 0, 0)
display.text(FsatUseDisplay, 190, 30, 240, 4)
print(FsatUseDisplay)

# TOP RIGHT, PART 3, VALID OR WARN STRING
display.set_pen(FfixTypeStrBbg)
display.text(FfixTypeStrB, 185, 60, 240, 2)
print(FfixTypeStrB)
print(FspeedMPHStr)

display.set_pen(255, 255, 255)
#BOTTOM - CURRENT SPEED
display.text(FspeedMPHStr, 0, 80, 240, 8)
print(FspeedMPHStr)

#display.text(str(counter), 200, 80, 240, 2)
    
display.update()
print("Completed Updating Display")`

nickwebcouk, 2021-03-21:
Honestly - just like that I figured it out. I hadn't converted FspeedMPHStr to a string and it was crashing.

Would it be better to have update() check to ensure STR's are passed or convert them - or at least warn about them on the console?

Gadgetoid, 2021-03-22:
I would guess if it's `FspeedMPHStr` not being converted into a `str` then it would fail at `display.text(FspeedMPHStr, 0, 80, 240, 8)` rather than `display.update()`, since the update just squirts the internal display buffer out to the attached device.

`display.text` handles the parsing of the input string, rendering it char-by-char into the internal buffer... will have to throw together a test and see what might have gone wrong here.

If you can reproduce with just a few lines of Python I'd be interested to know! Thank you.

SimUKdev, 2021-03-23:
Possibly similar to issue #58 ?

closed 2021-03-30 by Gadgetoid

Gadgetoid, 2021-03-30:
The hard-lock bug with passing a non-string type into `.text()` has been squashed into a `TypeError`- give v0.1.0 a try: https://github.com/pimoroni/pimoroni-pico/releases/tag/v0.1.0