pimoroni/pimoroni-pico #108: picodisplay.set_backlight() seems broken ...

teuler, 2021-03-29:
... it does not change the brightness of the display, while the following code does:

```
from machine import Pin, PWM
p = PWM(Pin(20))
p.freq(1000)
p.duty_u16(int((2**16)/2))
p.duty_u16(int((2**16)/4))
p.duty_u16(int((2**16)/8))
```

Tohaker, 2021-03-29:
On the latest master, I get this build error:
```
[ 64%] Building C object CMakeFiles/snake.dir/home/pi/pico/pico-sdk/src/rp2_common/hardware_timer/timer.c.obj
/home/pi/pico/snake/main.cpp: In function 'int main()':
/home/pi/pico/snake/main.cpp:164:19: error: 'class pimoroni::PicoExplorer' has no member named 'set_backlight'; did you mean 'set_clip'?
     pico_explorer.set_backlight(100);
                   ^~~~~~~~~~~~~
                   set_clip
```

So it looks like it may not even exist anymore...

Gadgetoid, 2021-03-31:
My apologies, it was/is broken due to #112

closed 2021-03-31 by teuler