pimoroni/pimoroni-pico #116: Build fails on RPi3

graeme-winter, 2021-04-04:
Building on RPi3 (freshly built, with functioning pico sdk etc.) build fails:

```
Scanning dependencies of target elf2uf2
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[ 11%] No install step for 'ELF2UF2Build'
[ 11%] Completed 'ELF2UF2Build'
[ 11%] Built target ELF2UF2Build
Scanning dependencies of target roundlcd_demo
[ 11%] Building CXX object examples/breakout_roundlcd/CMakeFiles/roundlcd_demo.dir/demo.cpp.obj
/home/pi/pico/pimoroni-pico/examples/breakout_roundlcd/demo.cpp: In function 'int main()':
/home/pi/pico/pimoroni-pico/examples/breakout_roundlcd/demo.cpp:49:25: error: request for member '_private_us_since_boot' in 'at', which is of non-class type 'absolute_time_t {aka long long unsigned int}'
         uint64_t t = at._private_us_since_boot / 100000;
                         ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [examples/breakout_roundlcd/CMakeFiles/roundlcd_demo.dir/build.make:63: examples/breakout_roundlcd/CMakeFiles/roundlcd_demo.dir/demo.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:2156: examples/breakout_roundlcd/CMakeFiles/roundlcd_demo.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

 I note also error in where to get µpython from in docs - repo should be:
```
pi@raspberrypi:~/pico/micropython/ports/rp2 $ git remote -v
origin	git@github.com:micropython/micropython.git (fetch)
origin	git@github.com:micropython/micropython.git (push)
```

?

(the repo listed in the README does not exist)

Gadgetoid, 2021-04-05:
Ooof, yes that line *should* be:

```
uint64_t t = to_us_since_boot(at) / 100000;
```

I should *not* use members that are marked "private" with two different conventions :grimacing: 

Will raise a PR to patch it.

closed 2021-04-05 by Gadgetoid

graeme-winter, 2021-04-05:
👍

Thanks for the quick fix

PR #117 (2021-04-05): Fix use of private variable in round LCD demo
Fixes #116
commit 3c66b2bc (2021-04-05): Fix use of private variable in round LCD demo

Fixes #116