pimoroni/pimoroni-pico #118: How to contribute?

simonhowkins, 2021-04-05:
I've got a small change to one of the examples that I'd like to contribute. How do I do that?

It looks like github has a standard way to document this - see https://github.com/pimoroni/pimoroni-pico/community

tramcrazy, 2021-04-11:
You would need to create your own branch, make the changes, and create a pull request describing the changes. Then someone will hopefully have a look at it and potentially merge it to the main repository.

Gadgetoid, 2022-01-25:
Contributing guidelines might be a good way to address #223

Though I think the broader process of forking/editing and raising a PR might be outside the scope of such things- there must be a good existing guide for that we can link to somewhere. This might be a little dry? https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

closed 2024-05-30 by thirdr