pimoroni/pimoroni-pico #119: `picodisplay.init(buf)` affects pin 9?

teuler, 2021-04-06:
Hi,
I am a bit puzzled: I am using 3 pins (2, 9, and 10) to drive servos via PWM. This works fine until I call `picodisplay.init(buf)`:

```
[...]
dx = picodisplay.get_width()
dy = picodisplay.get_height()
buf = bytearray(dx *dy *2)
picodisplay.init(buf)
```
Then the servo at pin 9 cannot be controlled any more and shows erratic behavior (e.g. drifts in one direction). This holds on until the rp2 is hard-resetted. It does not seem to matter if I call `picodisplay.init(buf)` before or after I grab pin 9 for the servo ... The pin "reports" still the correct frequency and duty cycle, but does not respond.

I looked at the underlying C/C++ code (https://github.com/pimoroni/pimoroni-pico) but cannot find any undocumented reference to pin 9. I used the newest code to built the image.

Does anyone have an idea what happens here?
Thanks!

teuler, 2021-04-06:
P.S.: I soldered a bridge on my PCB to use pin 21 instead of pin 9 and now it works for me. Still curious, this side effect on pin 9 ...

Gadgetoid, 2021-04-07:
I'm going to hazard a guess that pin 9 is part of the same PWM slice as pins 6, 7 and 8 used for the RGB LED. Or, specifically, pin 8.

[image]

This might be worth documenting :thinking:

teuler, 2021-04-07:
Yes, this sounds like a possible explanation. This potential issue is not clearly pointed out in the pico board description - at least, I did not find it. And yes, a warning - ideally in the rp2040 specific pages in the MicroPython docs - would be useful.

closed 2021-04-07 by teuler