pimoroni/pimoroni-pico #124: 1.3" SPI Colour Round LCD (240x240) Breakout does not work on Rpi pico

metanav, 2021-04-14:
I have compiled and flashed the breakout_roundlcd from the examples (C++) to Rpi Pico successfully. I have followed the pin map from the drivers/st7789/st7789.hpp. After flashing and resetting the display does not show anything. When I press reset button the display shows some colorful still graphics momentarily. I believe connection is OK since it displays something on keep pressing RESET button (Captain Resetti).

metanav, 2021-04-19:
With the same pin mappings as stated above it works by adding
` display.set_backlight(225);` after display.init();
I guess PWM gets activated only after set_backlight() call.

Gadgetoid, 2021-04-19:
Haha! I wish I'd seen this 35 minutes ago because I just had this revelation after *much* head scratching and updated #128 

I have added `set_backlight` to the example *and* into the library as a default value to avoid this surprise in future!

Thank you.

closed 2021-04-19 by metanav