pimoroni/pimoroni-pico #146: Wireless Crashes Display

tonygo, 2021-05-12:
Hi Phil
Just tried to run Display Workout program with Display, Wireless module and Pico on Pico Decker. Pressing either button A crashes the display screen.

https://www.instructables.com/Pimoroni-Pico-Display-Workout/

Can we please have the pinout for the wireless board and to which pins the track cuts points are connected?

Gadgetoid, 2021-11-26:
Wireless at least is documented on the store:

[image]

And while there's no graphic for Pico Display 2.0. I believe it's identical to the smaller Display Pack:

[image]

I'm not sure which of Pico Wireless's pins are essential, but it would seem the two could possibly work together with some gentle persuasion since they seem to have discrete chip-select pins.

Crucial problems are:

1. Display backlight overlaps the SD card data pins
2. Display RGB LED overlaps the ESP32 chip-select pin and (unused for normal use) the UART RX pin.

Button A does not appear to overlap, however, what code are you running?

Gadgetoid, 2021-11-26:
Okay you're not going to like this very contrived tangent, but I've been toying with a similar problem today and decided to go a little off piste. In short, you can use a separate Pico board to run Pico Wireless and then just hook up a couple of UART wires between the two.

On my transmitting side I have:

```python
uart = machine.UART(0, 115200, tx=machine.Pin(0), rx=machine.Pin(1), bits=8)
uart.write("{},{},{}\n".format(r, g, b))
```

And on my receiving side (due to bugs in Pico SDK that apparently totally break "readline") I have:

```python
uart = machine.UART(1, 115200, tx=machine.Pin(20), rx=machine.Pin(21), bits=8)

available = uart.any()
if available > 0:
    result = uart.read(available)
```

This might strike you at first as complete lunacy, but the memory requirements of Pico Display and Pico Wireless combined are probably going to make it quite a challenge to get the two working nicely on the same RP2040 anyway. At least in MicroPython where you don't have the most headroom. Chucking an extra (cheap as chips) Pico at the problem makes it really straight-forward.

I used this technique to very minimally modify our Python CheerLights example and get it to feed RGB triplets over UART to our matrix driver:

[image]
[image]

I do wish I could run two instances of Thonny, though! :grimacing: 

Yes two RP2040s and an ESP32 is an absurd cluster of micros to throw at a simple problem that should be possible to solve with one. But that would require *effort* and I'll have none of that sort of business around here :laughing:

closed 2024-03-11 by Gadgetoid