pimoroni/pimoroni-pico #150: Micropython + pico_display: Using LED with buttons.py causes crash

matthewsellis, 2021-05-14:
Took your buttons.py example, added 'display.set_led(0, 255, 255)' commands (with numbers to match text colour in each block), it reacts to the first button press then hangs with a blank screen.

Sample code:
```
# This example shows you a simple, non-interrupt way of reading Pico Display's buttons with a loop that checks to see if buttons are pressed.

import picodisplay as display
import utime

# Initialise display with a bytearray display buffer
buf = bytearray(display.get_width() * display.get_height() * 2)
display.init(buf)
display.set_backlight(0.5)

# sets up a handy function we can call to clear the screen
def clear():
    display.set_pen(0, 0, 0)
    display.clear()
    display.update()

while True:
    if display.is_pressed(display.BUTTON_A):              # if a button press is detected then...
        display.set_led(255, 255, 255)                    # Set LED to a converted HSV value
        clear()                                           # clear to black
        display.set_pen(255, 255, 255)                    # change the pen colour
        #display.set_pen( white )                    # change the pen colour
        display.text("Button A pressed", 10, 10, 240, 4)  # display some text on the screen
        display.update()                                  # update the display
        utime.sleep(1)                                    # pause for a sec
        clear()                                           # clear to black again
    elif display.is_pressed(display.BUTTON_B):
        clear()
        display.set_pen(0, 255, 255)
        #display.set_led(0, 255, 255)
        display.text("Button B pressed", 10, 10, 240, 4)
        display.update()
        utime.sleep(1)
        clear()
    elif display.is_pressed(display.BUTTON_X):
        clear()
        display.set_pen(255, 0, 255)
        #display.set_led(255, 0, 255)
        display.text("Button X pressed", 10, 10, 240, 4)
        display.update()
        utime.sleep(1)
        clear()
    elif display.is_pressed(display.BUTTON_Y):
        clear()
        display.set_pen(255, 255, 0)
        #display.set_led(255, 255, 0)
        display.text("Button Y pressed", 10, 10, 240, 4)
        display.update()
        utime.sleep(1)
        clear()
    else:
        display.set_led(0, 0, 0)
        display.set_pen(255, 0, 0)
        display.text("Press any button!", 10, 10, 240, 4)
        display.update()
    utime.sleep(0.1)  # this number is how frequently the Pico checks for button presses

```

helgibbons, 2021-05-19:
Hi Matthew! I'm getting the same with your code - I think it's the `set_led` in the else statement that's causing it to hard lock (though I'm not sure why - maybe because it's refreshing that part of the loop really quickly?).

Putting that instruction into the clear function instead seems to sort it :)

```# This example shows you a simple, non-interrupt way of reading Pico Display's buttons with a loop that checks to see if buttons are pressed.

import picodisplay as display
import utime

# Initialise display with a bytearray display buffer
buf = bytearray(display.get_width() * display.get_height() * 2)
display.init(buf)
display.set_backlight(0.5)

# sets up a handy function we can call to clear the screen
def clear():
    display.set_pen(0, 0, 0)
    display.set_led(0, 0, 0)
    display.clear()
    display.update()
  
while True:
    if display.is_pressed(display.BUTTON_A):              # if a button press is detected then...
        clear()                                           # clear to black
        display.set_pen(255, 255, 255)
        display.set_led(255, 255, 255)                    # change the pen colour
        display.text("Button A pressed", 10, 10, 240, 4)  # display some text on the screen
        display.update()                                  # update the display
        utime.sleep(1)                                    # pause for a sec
        clear()                                           # clear to black again
    elif display.is_pressed(display.BUTTON_B):
        clear()
        display.set_pen(0, 255, 255)
        display.set_led(0, 255, 255)
        display.text("Button B pressed", 10, 10, 240, 4)
        display.update()
        utime.sleep(1)
        clear()
    elif display.is_pressed(display.BUTTON_X):
        clear()
        display.set_pen(255, 0, 255)
        display.set_led(255, 0, 255)
        display.text("Button X pressed", 10, 10, 240, 4)
        display.update()
        utime.sleep(1)
        clear()
    elif display.is_pressed(display.BUTTON_Y):
        clear()
        display.set_pen(255, 255, 0)
        display.set_led(255, 255, 0)
        display.text("Button Y pressed", 10, 10, 240, 4)
        display.update()
        utime.sleep(1)
        clear()
    else:
        display.set_pen(255, 0, 0)
        display.text("Press any button!", 10, 10, 240, 4)
        display.update()
    utime.sleep(0.1)  # this number is how frequently the Pico checks for button presses
```

Gadgetoid, 2021-05-20:
The first example runs fine for me on the latest release... except that BUTTON_X and BUTTON_Y don't work for me.

We still have unresolved memory issues with the display products that are likely to cause all sorts of weird and unpredictable outcomes.

helgibbons, 2022-07-11:
Closing as this relates to now resolved underlying memory issues (feel free to reopen if you're still having probs though!)

closed 2022-07-11 by helgibbons