pimoroni/pimoroni-pico #158: Who wrote the sample code??? (rgb_keypad)

MikeMitterer, 2021-05-21:
Lately I bought a pi pico and I tried the sample code for the rgb_keypad (Python + C++)

Take this snipped from https://github.com/pimoroni/pimoroni-pico/blob/main/examples/pico_rgb_keypad/demo.cpp#L36-L47
```c++
  uint8_t button = 0;
  for(uint8_t find = 0; find < pico_keypad.NUM_PADS; find++) {
    // check if this button is pressed and no other buttons are pressed
    if(button_states & 0x01) {
      if(!(button_states & (~0x01))) {
        lit |= 1 << button;
      }
      break;
    }
    button_states >>= 1;
    button++;
  }
```
A for-loop with "find" which is never used but a variable "button" that is incremented inside of for - WHY???
Pure documentation, spaghetti code and a lot of shift operations - OMG. Please write better samples and document your code.

Gadgetoid, 2021-05-24:
This looks like some incredibly hastily written code for production testing that somehow got swept up into this repository. It's... :cry:  beautiful.

I specially like how `last_button_states = button_states;` happens both before *and* after `button_states` is right-shifted to oblivion.

This example is presumably preserved here so we get free publicity from r/programmerhumour and thedailywtf.

lowfatcode, 2021-05-24:
I'm sorry to say that I did. :-)

There was a lot going on for the Pico launch and you're right that some improvements can be made here, we'll give it another good look over!

bennuttall, 2021-05-24:
Saying "Who wrote this code?", "WHY???" and "OMG" is a poor way to engage and I'd hate to be on the receiving end of such harsh criticism. Play nice, @MikeMitterer. Respectfully, a community member who this sometimes happens to.

lowfatcode, 2021-05-24:
To be clear I can see @bennuttall 's point and definitely consider it the correct way to approach an issue report like this.

However, since I know the context in which this code was written (16 hour days for weeks on end...) it's like water off a ducks back - I'm not offended ;-D We'll get it fixed up, thanks for pointing it out!

Gadgetoid, 2021-05-24:
I rewrote that entire snippet to:

```c++
lit_buttons |= pico_keypad.get_button_states();
```

Doesn't have the femtoseconds of debounce that the extra instructions added, but I feel it captures the gist of it.

Gadgetoid, 2021-05-24:
@bennuttall this you?
[image]

PR #160 (2021-05-26): Rewrite this code for #158
Rewrites the cursed Pico RGB Keypad code so that it can't hurt anyone else.

Turns out the whole loop could have been a bitwise operator.

The true lesson was the people we met along the way.

Fixes #158
closed 2021-05-26 by Gadgetoid