pimoroni/pimoroni-pico #162: breakout_as7262 init fails since moving to common i2c

knightinlet, 2021-06-03:
Prior to moving to common i2c the breakout_as7262 example worked perfectly, but now seems to fail during init call.

knightinlet, 2021-06-04:
Apologies my bad!  When looking at the example again I see you now need to specify the board type in the I2C constructor. The example is set to BREAKOUT_GARDEN, but as I'm using the Pico Explorer I needed to change the value to PICO_EXPLORER which I have now done and it works fine.  So no issue. Thanks.

knightinlet, 2021-06-04:
Has anyone got the pico_pot_explorer example working on the latest version?  Seems to be a conflict between pico_explorer and breakout_potentiometer.  Comment either the pico_explorer or the breakout_potentiometer calls out and they work individually, but not at the same time.

Gadgetoid, 2021-06-04:
> Apologies my bad! When looking at the example again I see you now need to specify the board type in the I2C constructor. The example is set to BREAKOUT_GARDEN, but as I'm using the Pico Explorer I [...]

Music to my ears. The common I2C shift was to solve this exact problem because I got caught out over and over again by using the wrong pins on the wrong board.

I wonder if explorer and the potentiometer breakout are fighting over a common pin. Does the example outright crash?

Gadgetoid, 2021-06-04:
Just ran a quick test and I see the same thing happening. No time to fix it now (or, uh, potentially until the week after next) but there's definitely something awry here!

knightinlet, 2021-06-04:
Yes it just crashes. Prior to the common I2C shift the example worked. It came about as I was trying the BME680 breakout and PICO explorer. I could get each to work individually, but not at the same time so I went back to the examples and found the same issue with the pico_pot_explorer to make sure it wasn't something I was doing.   

To add to the mystery the pico_explorer example also fails now. It used to work with no intervention. I have to comment out the msa301.init() call and it starts working.  I suspect the root cause is the same.

knightinlet, 2021-06-04:
Thanks for the update. Glad you were able to recreate it and I wasn't just me going mad.

Gadgetoid, 2021-06-05:
Been poking this when I can grab a half hour now and then, and finally found and fixed in #165

The Pico Explorer library was trying to use pin 20 for a non-existent backlight connection, which coincided with the SDA pin for Pico Explorer's I2C bus. As you might imagine that breaks I2C pretty handily- forcing it into a perpetually locked state.

Explains the weird pulse I was seeing on the logic analyzer!

knightinlet, 2021-06-05:
Thanks for taking the time to investigate. I've just synced with the latest version and Pico_explorer,  as7262_explorer_bargraph and pico_pot_explorer examples are all fully functioning again :) Thanks again.