pimoroni/pimoroni-pico #168: Pico Explorer: SPI Breakouts

davll, 2021-06-06:
In Pico Explorer, SPI's MISO Pin and LCD's DC Pin are the same (GPIO 16), I am wondering if I can use SPI breakouts with it?

Gadgetoid, 2021-06-06:
It won’t be easy because the D/C pin is set up in GPIO mode- so you’ll have to switch it between GPIO and SPI on the fly to talk to a full duplex sensor. Probably easier just to set up another SPI bus on some spare pins.

davll, 2021-06-06:
Unfortunately, GPIO 0~7 are only for SPI 0, and no available pins for SPI 1...
Things would be even easier if the TFT LCD used SPI 1 with GPIO 8~15. :P

Gadgetoid, 2022-01-26:
I (and, sorry for the many moons delay between responses) wonder if an example showing SPI setup on those pins using PIO would be helpful? Or even bit-banged. I know it uses precious resources (PIO), or is slow (bitbanged), but it gets you somewhere at least.

Gadgetoid, 2024-03-11:
Closing as stale/complete.

closed 2024-03-11 by Gadgetoid