pimoroni/pimoroni-pico #169: Example use of interrupt for LTR559 breakout

damienh, 2021-06-09:
I have been getting started with the LTR559 breakout with my Pico, via the Explorer base, and would love to see if it's possible to get an example of how the interrupts could be used with the Pico. I found this example https://github.com/pimoroni/ltr559-python/blob/master/examples/proximity-interrupt.py for the `ltr559-python` library but it appears to be fairly different to the breakout-ltr559 library meant for the Pico.

Loving the products for the Pico. Explorer base has been really good fun to use.

Gadgetoid, 2022-01-25:
The threshold for Light isn't very intuitive since it's given in raw counts and not the Lux value. We should probably figure out how to convert that.

Anyway,  here's an example that sets up and uses the interrupt on an LTR-559 plugged into a Breakout Garden base. Adjust pins as necessary for Explorer

```python
import time
from machine import Pin
from pimoroni_i2c import PimoroniI2C
from breakout_ltr559 import BreakoutLTR559

PINS_BREAKOUT_GARDEN = {"sda": 4, "scl": 5}
PINS_PICO_EXPLORER = {"sda": 20, "scl": 21}
PIN_INTERRUPT = 22  # 3 for Breakout Garden

i2c = PimoroniI2C(**PINS_PICO_EXPLORER)
ltr = BreakoutLTR559(i2c, interrupt=PIN_INTERRUPT)
interrupt = Pin(PIN_INTERRUPT, Pin.IN, Pin.PULL_DOWN)

ltr.light_threshold(0, 10)  # COUNTS, NOT LUX!!!
ltr.proximity_threshold(0, 10)

def read(pin):
    reading = ltr.get_reading()
    if reading is not None:
        print("T: ", time.ticks_ms(), " Lux: ", reading[BreakoutLTR559.LUX], " Prox: ", reading[BreakoutLTR559.PROXIMITY])

interrupt.irq(trigger=Pin.IRQ_RISING, handler=read)

part_id = ltr.part_id()
print("Found LTR559. Part ID: 0x", '{:02x}'.format(part_id), sep="")

while True:
    pass
```

You don't get any control over the polarity of the interrupt pin (we could probably stand to add that) so you set it up as an input, pulled down and trigger an irq on the rising edge.

Gadgetoid, 2024-03-11:
Closed as complete.

closed 2024-03-11 by Gadgetoid