pimoroni/pimoroni-pico #172: Pico Display balls example seems to need the LEDs in Python?

ZontarOne, 2021-06-18:
Hello,
I don't know where to write this issue..  
I have bought the excellent Pico Display and I am trying to wire it by hand using a breadbord, and seeing which pins on the Pico I can use for other uses. 
I've used the pinout on this page.
https://shop.pimoroni.com/products/pico-display-pack

I've connected the Pico Display to the breadboard via some double headers, and from there to a raspberry Pico.

When I have the microPython balls example running, in order to see something on the display I need to have at least one led pin connected  from the DisplayHAT to the Pico (i mean, the RGB led pins on 9/10/11).
If I have a pin connected, the balls example run. Any pin will do.
As soon as I disconnect the LED pin (from the breadbord to the Pico) the demo stops.
If I try to run it again, it runs for about half a seconds and then freezes

on the console I get
`
Connection lost (read failed: [Errno 6] Device not configured)`

BTW i've tried the C++ demo, it does not seems to be affected even though it has a graphical glitch.

Gadgetoid, 2021-06-23:
This would suggest that Pico Display is using that one LED pin as a ground return path, since these are (probably I can't recall) pulled low. That's... not optimal! You need to fix your ground connections.

Do you have any ground connections wired up? Due to the Pico Display PCB being designed with the assumption it's connected to *all* of the ground pins on your Pico you might need anywhere between one and all of the ground pins connected for it to function properly.

ZontarOne, 2021-06-23:
Hello, 
it might stand to reason, and in fact no, I don't have all GND pins to ground, because I've checked the minimum number of ground needed for the display, but of course one or more is needed in this setup.
What puzzles me is that the same demo, same pin connections, in "C" works a9lbeit with the grapihcs twitch that some other user has mentioned), in Pyhton shows this strange behaviour that freezes

BTW this is just a curiosity, I use C and at the moment I use the display to show video fteched by an external camera
Thanks

closed 2021-06-23 by ZontarOne