pimoroni/pimoroni-pico #175: SPI LCD pins seem inconsistent with pimoroni website

steven-dawkins, 2021-07-03:
I have been struggling to get the 1.54" SPI colour lcd working on a pi pico using micropython

https://shop.pimoroni.com/products/1-54-spi-colour-square-lcd-240x240-breakout

These are the instructions from the pimoroni website.
```
Here's which pins to connect between your LCD breakout and your Pi's GPIO (note that it's BCM pin numbering):

    3-5V to any 5V or 3V pin
    CS to BCM 7
    SCK to BCM 11
    MOSI to BCM 10
    DC to BCM 9
    BL to BCM 19
    GND to any ground pin
```

I am a bit perplexed by the BCM numbering and I believe these change between various pi devices, I guess there is a pi device where the above pins make sense but since these instructions are meant for different pi's I found this quite misleading.

Fortunately the Breakout class reports what GPIO pins it is using so I was able to discover the correct pins to get it working. I don't think it is possible to work out the correct pins for a pico using the BCM pins listed on the website.

```
> print(display)
BreakoutColourLCD240x240(spi = 0, cs = 17, dc = 16, sck = 18, mosi = 19, bl = 20)
```

It might be useful to add something along these lines to the demo code to give people some guidance in the demo output what the correct wiring should be.
https://github.com/pimoroni/pimoroni-pico/blob/5d27dbc7119f4ded9d58aa628a673bf13dc3ebb0/micropython/examples/breakout_colourlcd240x240/demo.py

Gadgetoid, 2021-07-04:
The BCM pin numbering is for Raspberry Pi single-board computers only, and doesn't apply to Pico at all.

On Pico it's assumed you'll use a Breakout Garden board and plug into one of the SPI slots, the pins for which are here: https://github.com/pimoroni/pimoroni-pico/blob/37d86ee766b15ce7bc57b281f1a6b6f081223f2a/common/pimoroni_common.hpp#L21-L30

But you can, of course, wire the LCD manually to those pins.

helgibbons, 2021-08-04:
I've added Pico wiring info for the LCD breakouts to the shop pages, hopefully this should make it a little easier to find!