pimoroni/pimoroni-pico #176: RV3028 RTC with PICO, manual set time doesn't work correctly.

alphanumeric007, 2021-07-07:
Pi Pico with latest Pimoroni uf2 file, attached to PICO Breakout Garden with RV3028 in first i2c socket.

rtc.set_time(00, 00, 09, 02, 06, 07, 00)
gets me
Date: 06/07/2048, Time: 09:00:20AM
and
rtc.set_time(00, 00, 09, 02, 06, 07, 21)
gets me
Date: 06/07/2069, Time: 09:00:00AM

Using 2021 for year gets me
ValueError: year out of range. Expected 0 to 99

EDIT: Is the following correct?
rtc.set_time(sec, min, hour, weekday, date, month, year)
sec = 0-59
min = 0-59
hour = 0-23
weekday = 1-7
date = 1-31
month = 1-12
year = 0-99

Somebody else on the Pimoroni forum suggested the following.

It looks like a library bug. rtc.set_time constrains the year to the range [00-99], but then calls the corresponding RV3028 driver routine, which treats it as a 16-bit unsigned int and subtracts 2000 from it.

When you take all that into account, the results you got are actually inevitable:

00 (rtc) → -2000 (RV3028) → 0xf830 (16-bit uint) → 72 (BCD) → 48 (decimal)

21 (rtc) → -1979 (RV3028) → 0xf845 (16-bit uint) → 105 (BCD) → 69 (decimal)

2021 (rtc) → 21 (RV3028) → 0x0015 (16-bit uint) → 33 (BCD) → 21 (decimal)

Methinks rtc.set_time should actually take the full year instead of artificially constraining the input value.

https://forums.pimoroni.com/t/rv3028-rtc-with-pico-set-time-to-local-time/17419

Gadgetoid, 2021-07-07:
Looks like you - or the forum user at least - are correct. I definitely prefer the "2000 to 2099" range exposed by the RV3028 library, so I'll tweak Python to enforce that instead.

Gadgetoid, 2021-07-07:
It *should* be fixed in this CI build of MicroPython if you want to give it a shot (and if this link actually works for you!) - https://github.com/pimoroni/pimoroni-pico/suites/3180769646/artifacts/73343152

Gadgetoid, 2021-07-07:
I am not smart.

alphanumeric007, 2021-07-07:
Thanks Phil, I downloaded and loaded that firmware.uf2 file to that PICO and the set time now works. Entering 2021 for year got me 2021.

closed 2021-07-07 by alphanumeric007