pimoroni/pimoroni-pico #184: Pico Display Issue, Backlight unable to set.

JamieH93, 2021-07-24:
I am trying to use multiple addon board and when I call the RTC function from Waveshare(https://thepihut.com/products/precision-rtc-module-for-raspberry-pi-pico-ds3231), It makes the backlight 100%.

I'm trying to figure out if it's a pin conflict or something deeper.
Currently I'm using a Waveshare RTC + UPS.
They are stacked - Pico, UPS, RTC, Display.

I've used the buttons example code as my base, The below code is called when I press my Y button.

`
`bus = I2C(0,scl=Pin(21),sda=Pin(20))

    address = 0x68
    start_reg = 0x00   
    w  = ["Sunday","Monday","Tuesday","Wednesday","Thursday","Friday","Saturday"]
    t = bus.readfrom_mem(int(address),int(start_reg),7)
    a = t[0]&0x7F  #second
    b = t[1]&0x7F  #minute
    c = t[2]&0x3F  #hour
    d = t[3]&0x07  #week
    e = t[4]&0x3F  #day
    f = t[5]&0x1F  #month
    display.text("%02x/%02x/20%x %02x:%02x:%02x %s" %(t[4],t[5],t[6],t[2],t[1],t[0],w[t[3]-1]),10,0,240,4)``

Pinkyatacme, 2021-08-08:
I think your pin definitions are the issue:
  bus = I2C(0,scl=Pin(21),sda=Pin(20))
According to the schematics of the Waveshare RTC board, you should be using different pins (6 for SCA, and 7 for SCL). GP20 is pin26 of the Pico, which is used for the BL_EN signal on the Display Pack, so you're pumping your I2C data into the backlight control line!

JamieH93, 2021-08-09:
> I think your pin definitions are the issue:
> bus = I2C(0,scl=Pin(21),sda=Pin(20))
> According to the schematics of the Waveshare RTC board, you should be using different pins (6 for SCA, and 7 for [...]

Those are for the old version of the board.

[image]

So it's a pin conflict as suspected, Shall buy a different screen, Cheers!

closed 2021-08-09 by JamieH93