pimoroni/pimoroni-pico #185: Drawing Images on the Display Pack in Python

c-gaskell, 2021-07-24:
I was wondering if there was any way to draw an image file on the display using Python? I couldn't see anything in the docs.

Gadgetoid, 2021-08-02:
It's possible - in theory - but I can't seem to find any good, existing way to deal with image formats on MicroPython. The solution might be to convert your source image into raw bytes (packed into RGB565 2-bytes per pixel format) - something desktop Python/PIL can help with - and then replace the display buffer with the result.

Here's a basic example of a conversion script. Run this on your computer supplying the filename of an image you want to convert:

```python
#!/bin/env python3
from PIL import Image
import numpy
import sys
import pathlib

# Run with `./filename.py source-image.jpg`
IMAGE_PATH = pathlib.Path(sys.argv[1])
OUTPUT_PATH = IMAGE_PATH.with_suffix(".16bpp")

# Pico Display is 240x135 pixels
WIDTH = 240
HEIGHT = 135

def image_to_data(image):
    """Generator function to convert a PIL image to 16-bit 565 RGB bytes."""
    # NumPy is much faster at doing this. NumPy code provided by:
    # Keith (https://www.blogger.com/profile/02555547344016007163)
    pb = numpy.array(image.convert('RGB')).astype('uint16')
    color = ((pb[:, :, 0] & 0xF8) << 8) | ((pb[:, :, 1] & 0xFC) << 3) | (pb[:, :, 2] >> 3)
    return numpy.dstack(((color >> 8) & 0xFF, color & 0xFF)).flatten().astype('uint8').tobytes()

img = Image.open(IMAGE_PATH)
img = img.resize((WIDTH, HEIGHT))
data = image_to_data(img)

print(f"Converted: {len(data)} bytes, Expected: {WIDTH * HEIGHT * 2} bytes.")

with open(OUTPUT_PATH, "wb") as f:
    f.write(data)

print(f"Written to: {OUTPUT_PATH}")
```

And then to display the image, read it into Pico Display's buffer like so:

```python
import utime
import picodisplay as display

IMAGE_PATH = "cat-and-bubble.16bpp"

# Set up and initialise Pico Display
buf = bytearray(display.get_width() * display.get_height() * 2)
display.init(buf)
display.set_backlight(0.8)

print("Reading: {}".format(IMAGE_PATH))

open(IMAGE_PATH, "rb").readinto(buf)

while True:
    display.update()          # Update the display
    utime.sleep(1.0 / 60)

```

Here's a quick example using the first 240x135 pixel image I could lay my hands on:

[image]

c-gaskell, 2021-08-03:
That looks great, and in theory memory shouldn't be an issue right? Since each image would only be around 60KB.

Also, is there a way you could draw the image onto an existing framebuffer? So I can integrate an image into a UI.

Gadgetoid, 2021-08-03:
> Also, is there a way you could draw the image onto an existing framebuffer? So I can integrate an image into a UI.

There's no good way to do this at the moment, since via the API you end up setting the pen colour and drawing the image pixel-by-pixel.

To do it in the interim you'd need a function that copies a smaller image across to a portion of the raw framebuffer. And yes- we probably need to add this to the API!

Taking the `convert.py` above and modifying it to create a smaller image, I produced "s4m_ur4i-dingbads.16bpp" from one of our 32blit spritesheets - https://github.com/32blit/32blit-sdk/blob/master/assets/s4m_ur4i-dingbads.png

And then hacked up this mess to actually display a region from the spritesheet at a particular location on the screen:

```python
import utime
import picodisplay as display
import random

ICON_PATH = "s4m_ur4i-dingbads.16bpp"
ICONS_WIDTH = 128
ICONS_HEIGHT = 128
ICON_WIDTH = 8
ICON_HEIGHT = 8

# Set up a buffer for our icons
icon_buf = bytearray(ICONS_WIDTH * ICONS_HEIGHT * 2)

DISPLAY_WIDTH = display.get_width()
DISPLAY_HEIGHT = display.get_height()

# Set up and initialise Pico Display
display_buf = bytearray(DISPLAY_WIDTH * DISPLAY_HEIGHT * 2)
display.init(display_buf)
display.set_backlight(0.8)z

# Load the icon data into the 128x128 (2 bytes per pixel) buffer
print("Reading: {}".format(ICON_PATH))
open(ICON_PATH, "rb").readinto(icon_buf)

def blit_icon(src, dst, src_rect, dest_point):
    src_x, src_y, src_w, src_h = src_rect
    dst_x, dst_y = dest_point
    
    src_w *= 2  # Two bytes per pixel
    
    for y in range(src_h):
        src_offset = ((src_y + y) * ICONS_WIDTH) + src_x
        dst_offset = ((dst_y + y) * display.get_width()) + dst_x
        src_offset *= 2  # Two bytes per pixel
        dst_offset *= 2  # Two bytes per pixel
        dst[dst_offset:dst_offset + src_w] = src[src_offset:src_offset + src_w]

display.set_pen(50, 50, 80)
display.clear()
for x in range(300):
    icon_x = random.randint(0, 15) * ICON_WIDTH  # Icons are 8x8 pixels
    icon_y = random.randint(0, 15) * ICON_HEIGHT
    dest_x = random.randint(0, DISPLAY_WIDTH - ICON_WIDTH)
    dest_y = random.randint(0, DISPLAY_HEIGHT - ICON_HEIGHT)
    blit_icon(icon_buf, display_buf, (icon_x, icon_y, ICON_WIDTH, ICON_HEIGHT), (dest_x, dest_y))

while True:
    display.update()          # Update the display
    utime.sleep(1.0 / 60)
```

This all boils down to the `blit_icon` call which takes the buffer to copy the icon from, the buffer to copy it to, a tuple with x, y, w, h values for the source, and a tuple of x, and y values for the destination.

And... well these 8x8 icons are a bit small :laughing: 

[image]

Kinda needs pixel doubling, just in case.

helgibbons, 2022-07-11:
Things are much improved now with regards to displaying images - JPEGDEC lets you decode jpegs directly without faffing with raw bytes.

https://github.com/pimoroni/pimoroni-pico/blob/main/libraries/pico_graphics/README.md

closed 2022-07-11 by helgibbons