pimoroni/pimoroni-pico #189: v0.2.4 missing drivers

wmarsey, 2021-08-04:
Hi

Thanks for this repository!

The newest version seems to be missing some files. I've tested doing a `mkdir build` / `cmake ..` / `make` on both v0.2.3 and v0.2.4. v0.2.3 succeeds, but v0.2.4 errors on the cmake with:

```
CMake Error at examples/breakout_bme688/bme688_parallel.cmake:3 (add_executable):
  Cannot find source file:

    /Users/william/Pi/pimoroni-pico-2.4/drivers/bme68x/src/bme68x.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  examples/breakout_bme688/CMakeLists.txt:2 (include)

CMake Error at examples/breakout_bmp280/bmp280_basic.cmake:3 (add_executable):
  Cannot find source file:

    /Users/william/Pi/pimoroni-pico-2.4/drivers/bmp280/src/bmp280.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  examples/breakout_bmp280/CMakeLists.txt:1 (include)

CMake Error at examples/breakout_bme280/bme280_basic.cmake:3 (add_executable):
  Cannot find source file:

    /Users/william/Pi/pimoroni-pico-2.4/drivers/bme280/src/bme280.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  examples/breakout_bme280/CMakeLists.txt:1 (include)
```

and then fails on make with:

```
In file included from /Users/william/Pi/pimoroni-pico-2.4/examples/breakout_bme688/bme688_parallel.cpp:5:
/Users/william/Pi/pimoroni-pico-2.4/drivers/bme68x/bme68x.hpp:5:10: fatal error: bme68x.h: No such file or directory
    5 | #include "bme68x.h"
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [examples/breakout_bme688/CMakeFiles/bme688_parallel.dir/bme688_parallel.cpp.obj] Error 1
make[1]: *** [examples/breakout_bme688/CMakeFiles/bme688_parallel.dir/all] Error 2
```

wmarsey, 2021-08-04:
Oh scratch that -- it needed a `git submodule update --init` which I hadn't thought to try

So I think the issue is actually that https://github.com/pimoroni/pimoroni-pico/blob/main/setting-up-the-pico-sdk.md needs an update

Gadgetoid, 2021-08-04:
Ooof! Yes. I will get the documentation updated. Relied too much on the continuous integration telling me it worked and totally neglected that!

closed 2021-10-07 by Gadgetoid