pimoroni/pimoroni-pico #199: The PI PICO DISPLAY demo under C/C++, doesn't compile and don't work.

cmarcante, 2021-08-28:
Please, check that the PI  pico-display functions does not work under C,C++.
/* */
PROBLEM: The file pico_display.hpp includes the file -->libraries/pico_graphics/st7789/st7789.hpp<-- but the file "st7789.hpp" is in other folders.
The GIT-HUB has the file st7789.hpp in two places, but without the /pico_graphics/ folder.
// Place  #1 C:\GitHub\pimoroni\pimoroni-pico\drivers\st7789\st7789.hpp
// Place   #2 C:\GitHub\pimoroni\pimoroni-pico\libraries\st7789\st7789.hpp
This missing file, genarate an error to compile under C/C++.
/* */
PROBLEM: The compiler informs an error in st7789.hpp. Missing PIMORONI_SPI_DEFAULT_INSTANCE
Thanks

Gadgetoid, 2021-08-28:
There is no `st7789` in libraries? See - https://github.com/pimoroni/pimoroni-pico/tree/main/libraries

I think your local copy is in a bad state and you should probably tuck it to one side and clone anew.

cmarcante, 2021-08-29:
Thanks for your help.

I do not know why, I download the examples files from https://github.com/pimoroni/pimoroni-pico.git.

About the st7789.hpp file; we can see that in file -->pimoroni-pico/libraries/pico_display/pico_display.hpp<-- ;  the line number 3, there is an #include "drivers/st7789/st7789.hpp".

Thanks, I will clone again from the GITHUB; with the address provider by you and test again.

Thanks.

cmarcante, 2021-08-29:
Thanks for your help.

I do not know why, I download the examples files from https://github.com/pimoroni/pimoroni-pico.git.

Thanks, I will clone again from the GITHUB; with the address provider by you and test again.

Thanks.

cmarcante, 2021-08-29:
Hello,
Thanks, I cloned again, and the 3015 files were compiled OK with Visual Studio.

I will follow the advice from "Setting up the Pico SDK " and will try first; the example provided by PIMORI for PICO DISPLAY; **without any change**.

Regards

closed 2021-08-29 by cmarcante