pimoroni/pimoroni-pico #202: examples > plasma2040 > rotary > plasma2040_rotary.cpp function swap defined but not used

PaulskPt, 2021-09-08:
Ref: examples > plasma2040 > rotary > plasma2040_rotary.cpp

From line 34   is defined: float wrap(float v, float min, float max) { [...]}
but the function is not used. 
What was the idea to do with it?

ZodiusInfuser, 2021-09-10:
Hi. I cannot find a `wrap` function on line 34, but there is one on line 67 here: https://github.com/pimoroni/pimoroni-pico/blob/37209cd0c005c8c186fa63ab1c06ec714f2f525a/examples/plasma2040/plasma2040_rotary.cpp#L67
which gets called in the `colour_cycle` function on line 84.
https://github.com/pimoroni/pimoroni-pico/blob/37209cd0c005c8c186fa63ab1c06ec714f2f525a/examples/plasma2040/plasma2040_rotary.cpp#L84

Unless you were referring to another file I may have accidentally copied that function in to?

PaulskPt, 2021-09-10:
Hi, thank you for your reply. Excuse me, I must have been mistaken. I let the editor (Notepad++) do a 'find' for it. 
Yes, I see it now in the file currently in the repo here online. I'll close the issue.

closed 2021-09-10 by PaulskPt